arduino/arduino-builder #130: Not generating correct forward declarations for functions marked with extern C

pfeerick, 2016-03-30:
Just adding this an issue to be reviewed, as it arose when trying to get some code working with a third party core. It would appear that the arduino-builder does not generation declarations for functions marked with extern C. Code that worked with 1.6.5 (and I believe also 1.6.6) stopped working on 1.6.7 or 1.6.8 with c++ linkage errors as mentioned at https://github.com/esp8266/Arduino/issues/1759.

igrr ended up having to do the workaround committed at https://github.com/esp8266/Arduino/commit/730764075ea75e6c7f07f269367a8c92716d4c07 in order to allow the code to work on Arduino 1.6.8, and you can see from the commit the lines that were removed which worked prior to arduino-builder.

facchinm, 2016-03-30:
Hi @pfeerick , the issue should be solved in nightly as https://github.com/arduino/arduino-builder/pull/128 was merged. Also, @igrr already reported it here https://github.com/arduino/arduino-builder/issues/126, so I believe I can close this one :smile:

closed 2016-03-30 by facchinm

pfeerick, 2016-03-30:
lol... no worries... no wonder nothing jumped out at me for it as it was a closed issue! So since this is in the nightly/hourly build, does that mean it should appear in the next IDE stable release?

facchinm, 2016-03-30:
Yep :smile: