arduino/arduino-builder #138: How do i supply build arguments

Pimmetje, 2016-04-29:
For CI purpose i want to be able to use something similar to 
`-DDEBUG=1`
that should create something like:
`#define DEBUG 1`
I would like this so i can make special builds in a CI server.
For more information see my question here:
http://arduino.stackexchange.com/a/23682/20548

How can i do this with arduino-builder?

Pimmetje, 2016-05-01:
I think this is done using: 
`-prefs=build.extra_flags=-DDEBUG=1`
Can anyone tell me if this is correct?

matthijskooijman, 2016-05-01:
@Pimmetje, yes, that seems correct to me.

Pimmetje, 2016-05-01:
Thanks. I tried it for my build script. It works for one arguement. But when i want to add two : 
`-prefs="build.extra_flags=-DDEBUG=1 -DDEBUG2=1"`
I get a error : flag provided but not defined: -DDEBUG2

Pimmetje, 2016-05-04:
Adding -verbose and -debug=10 does not make any difference as far as i can tell. Is there any way i can debug this, Or does anyone knows where this is going wrong?

matthijskooijman, 2016-05-04:
IIRC the debug option is -debug-level=10 and it needs to come *before* the filename. Perhaps that helps?

Pimmetje, 2016-05-04:
@matthijskooijman thanks for the anwser. Sadly i already did it right here is the full output: 
```
/home/gitlab-runner/tmp/arduino-1.6.8/arduino-builder -hardware /home/gitlab-runner/tmp/arduino-1.6.8/hardware -tools /home/gitlab-runner/tmp/arduino-1.6.8/hardware/tools/avr -tools /home/gitlab-runner/tmp/arduino-1.6.8/tools-builder -libraries /home/gitlab-runner/tmp/arduino-1.6.8/libraries -libraries /home/gitlab-runner/tmp/arduinolibs -fqbn arduino:avr:mega:cpu=atmega2560 -build-path /home/gitlab-runner/tmp/113-66b5ed4d56ea35971bd0f2ffaa3c0a0b169b3a70 -debug-level=10 -verbose -dump-prefs -prefs="build.extra_flags=-DDEBUG=1 -DDEBUG2=1" /home/gitlab-runner/builds/fb8986ba/0/Arduino/Stopcontacten/Stopcontacten.ino
flag provided but not defined: -DDEBUG2
Usage of /home/gitlab-runner/tmp/arduino-1.6.8/arduino-builder:
  -build-options-file="": Instead of specifying --hardware, --tools etc every time, you can load all such options from a file
  -build-path="": build path
  -built-in-libraries=[]: Specify a built-in 'libraries' folder. These are low priority libraries. Can be added multiple times for specifying multiple built-in 'libraries' folders
  -compile=false: compiles the given sketch
  -core-api-version="10600": version of core APIs (used to populate ARDUINO #define)
  -debug-level=5: Turns on debugging messages. The higher, the chattier
  -dump-prefs=false: dumps build properties used when compiling
  -fqbn="": fully qualified board name
  -hardware=[]: Specify a 'hardware' folder. Can be added multiple times for specifying multiple 'hardware' folders
  -ide-version="10600": [deprecated] use 'core-api-version' instead
  -libraries=[]: Specify a 'libraries' folder. Can be added multiple times for specifying multiple 'libraries' folders
  -logger="human": Sets type of logger. Available values are 'human', 'machine'
  -prefs=[]: Specify a custom preference. Can be added multiple times for specifying multiple custom preferences
  -preprocess=false: preprocess the given sketch
  -quiet=false: if 'true' doesn't print any warnings or progress or whatever
  -tools=[]: Specify a 'tools' folder. Can be added multiple times for specifying multiple 'tools' folders
  -verbose=false: if 'true' prints lots of stuff
  -version=false: prints version and exits
  -vid-pid="": specify to use vid/pid specific build properties, as defined in boards.txt
  -warnings="": Sets warnings level. Available values are 'none', 'default', 'more' and 'all'
```

matthijskooijman, 2016-05-13:
@Pimmetje, I just tried a similar commandline to yours, which works as expected:

```
$ arduino-builder -dump-prefs -hardware "/home/matthijs/.arduino15/packages"  -tools "/home/matthijs/.arduino15/packages"  -fqbn=arduino:samd:arduino_zero_native  -prefs="build.extra_flags=-DDEBUG=1 -DDEBUG2=1" |grep ^build.extra_flags                                                                                                                       build.extra_flags=-DDEBUG=1 -DDEBUG2=1
```

I stripped as much of the commandline as possible to make it easier to read, but I got the same result using the commandline the IDE uses with your extra flags option appended. Looking at the error, it seems like the `-prefs` option is somehow broken up and supplied as two separate arguments to arduino-builder. The quotes in the command should normally prevent that, as confirmed by what happens when I remove the quotes:

```
$ arduino-builder -dump-prefs -hardware "/home/matthijs/.arduino15/packages"  -tools "/home/matthijs/.arduino15/packages"  -fqbn=arduino:samd:arduino_zero_native  -prefs=build.extra_flags=-DDEBUG=1 -DDEBUG2=1 2>&1 | head -1
flag provided but not defined: -DDEBUG2
```

Perhaps your shell is not properly interpreting these quotes? What OS are you running and what shell? Are you running the command you pasted from a shell, or through some script or other means?

Pimmetje, 2016-05-13:
I an running it as bash script on a docker ubuntu14.04 image. As part of a CI script. I tried it manually but i could not figure it out. Ill try yours and see what it does.

Pimmetje, 2016-05-13:
It took me a hour to figure i can better move the quotes outside of the variable i am using. Now it all works. Thanks for the tip. Now i can use it to pass some information to my libraries :+1:

closed 2016-05-13 by Pimmetje

matthijskooijman, 2016-05-13:
Yeah, doing quoting properly in a bash script is a PITA. Good to hear you figured it out :-D