arduino/arduino-builder #139: Arduino builder compiles hidden directories

NicoHood, 2016-05-01:
I sometimes need to move my code somewhere else to back it up for a huge api change. I wish I could use a `.old` folder for this, but arduino builder still compiles those files.

Example error:
`/Arduino/sketchbook/libraries/IRLremote/src/.old/IRLremoteReceive.hpp:141:29: error: 'dataNec' was not declared in this scope`

I guess the fix is quite simple, as only hidden folders has to be excluded from compiling/linking.

matthijskooijman, 2016-05-01:
I was actually working on a refactor of some of the compilation steps that actually changed exactly this. I'll make sure to reference this issue from my PR when I'm done :-)

cmaglie, 2016-05-23:
Fixed with https://github.com/arduino/arduino-builder/commit/405a24a7e3de1d48e1dc2b6bf152b1342a578b78

matthijskooijman, 2016-08-01:
Nope, this wasn't fixed by that.