arduino/arduino-builder #140: Missing prototype in Arduino 1.6.8

Lauszus, 2016-05-02:
It seems like Arduino 1.6.8 does not generate the function prototypes for the following examples in the USB Host shield library correctly: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/WiiMulti/WiiMulti.ino and https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/Bluetooth/PS3Multi/PS3Multi.ino.

More specifically the function prototype ```void onInit();``` does not seem to be properly generated.

This was originally reported by a user at the following issue: https://github.com/felis/USB_Host_Shield_2.0/pull/237.

A workaround for now is just to create the function prototype manually:
```C++
void onInit();
```

The same examples work fine in Arduino 1.6.7.

The full debug output is available here: https://gist.github.com/Lauszus/3daf5e6a5a6dde47b849b72f498dbdc9.

per1234, 2016-05-02:
The bug is caused by the function of the same name in the library: https://github.com/felis/USB_Host_Shield_2.0/blob/master/Wii.h#L430.
A simplified sketch that demonstrates the issue is:
```arduino
class test {
    void asdf() {}
};
void setup() {
  asdf();
}
void loop() {}
void asdf() {}
```
With Arduino IDE 1.6.8 and 1.6.9 2016/04/28 04:33 this fails to compile. With Arduino IDE 1.6.7 and previous it successfully compiles.

matthijskooijman, 2016-05-03:
I suspect #131 would "fix" the original bug (not the testcase @per1234 provided), by not running ctags on the library code, just on the sketch code.

However, looking at the ctags output, this is what is generated for the `asdf()` method in the last example:

asdf  sketch_may03a.ino   /^    void asdf() {}$/;"        kind:function   line:4  class:test      signature:()    returntype:void

The class:test indicates that this is a method, not a function, but I suspect arduino-builder ignores that, so that should be fixed.

@per1234, thanks for looking into this and pointing out the cause! @facchinm, would you look into fixing this?

facchinm, 2016-05-03:
I'm taking a look right now.
Indeed, the Ps3Multi issue is solved by #131 , but @per1234 's simplified example is not (of course). The builder needs to be aware of class tag and skip any function outside the main scope. I'll patch this adding a commit to #131

matthijskooijman, 2016-05-18:
#154 should fix this issue completely (both the original one and the testcase from @per1234).

Lauszus, 2016-05-18:
@matthijskooijman thanks and @per1234 thanks a lot for tracking down the cause of the bug!

closed 2016-08-26 by cmaglie

nitturis, 2019-02-19:
arduino 1.8.7 has same issue

2scorpio, 2020-12-27:
arduino 1.8.13 has same issue