arduino/arduino-builder #143: Class with same function name confuses arduino-builder

NicoHood, 2016-05-05:
Correct me if this is an IDE bug.

If a class exists with the same function name of a normal function it will throw an error. Very confusing. just try:
```cpp
void setup() {}

void loop() {
  rainbow();
}

class CTest {
  public:
    void rainbow() {
      Serial.println("test2");
    }

};

void rainbow() {
  Serial.println("test");
}
```

facchinm, 2016-05-05:
Hi Nico, good point, it is a side effect of fixing ctags between 1.6.7 and 1.6.8.
The bug is being tracked here https://github.com/arduino/arduino-builder/issues/140#issuecomment-216376704 and you can find a fix in this PR https://github.com/arduino/arduino-builder/pull/131 (if you could test it it would be great!).

closed 2016-05-05 by facchinm

NicoHood, 2016-05-05:
Bug resolved!