arduino/arduino-builder #159: arduino-builder fails if tools directory contains broken symlinks

lmoellendorf, 2016-06-05:
Steps to reproduce:

Create a tools dir with broken symlink:
`mkdir tools-dir`
`ln -s /no/way tools-dir/broken-link`

Call arduino-builider with created tools dir:
`~/Makeblock/arduino-1.6.9/arduino-builder -hardware "~/Makeblock/arduino-1.6.9/hardware/" -tools "~/tools-dir" -fqbn "arduino:avr:uno" -verbose ~/Arduino/mBot/mBot-default-program/mBot-default-
program.ino`

The error is:
`stat tools-dir/broken-link: no such file or directory`

Note, this error also occurs if a valid tools directory is specified which contains broken sym-links.

oqibidipo, 2016-06-09:
I was editing a file in my own library with Emacs and this happened:

`stat /Users/rjl/Documents/Arduino/libraries/_test/.#test_warning.h: no such file or directory`

For those not familiar with Emacs: that is the lock file it creates and it is a symlink to `<user>@<host>.<process_id>`.

facchinm, 2016-07-07:
Hi @lmoellendorf ,
could you test #166 and report if it solves your bug (and if it doesn't introduce any regression, of course :smile: )?

oqibidipo, 2017-02-25:
Not completely fixed: it still fails if there is a broken symlink directly under `.../libraries`.
Tested with 1.3.23, 1.3.24, PR213.

facchinm, 2017-02-27:
Ugh, are you sure about that? I tried to reproduce by creating a broken symlink in `$sketchbook/libraries` and the sketches still compile...

oqibidipo, 2017-02-27:
That depends on which libraries the sketch includes and where they are located: 
(target = Uno)

`#include <SomeInstalledLibrary.h>`
fails if a broken link is in `$skechbook/libraries`

`#include <SPI.h>`
fails if a broken link is in `hardware/avr/libraries`
compiles if a broken link is in `$skechbook/libraries`

`#include <Mouse.h>`
cannot find `"HID.h"` if a broken link is in `hardware/avr/libraries`

Clarification: 'fails' means that the directory in question is not added to the search path. The compilation may still succeed if the header can be found under a 'clean' directory.

facchinm, 2017-02-27:
Got it! I've been able to reproduce, reopening

facchinm, 2017-02-28:
Hi @oqibidipo , could you test #216 as soon as @ArduinoBot produces the build? Thanks!

oqibidipo, 2017-02-28:
Looks good.

akkana, 2017-03-17:
Saw this in 1.8.1 downloaded tonight. It wouldn't have been a problem if there had been an error message about the bad link, but since the only error message was a failure to include the .h file from a library that really was there, it took hours to track down the real problem.

facchinm, 2017-09-12:
Fixed by merging 		14ca8f2

commit 8bbac23d (2016-08-26): Avoid breaking the build if broken symlinks found

Fixes #159, please make sure that it doesn't introduce any regression

Tests are passing on Linux but something nasty could happen on different OS
closed 2017-09-12 by facchinm