arduino/arduino-builder #164: fatal error: ESP8266WiFi.h: No such file or directory

ashokelumalai, 2016-06-25:
I am stuck on this error

**## C:\Program Files (x86)\Arduino\libraries\blynk-library-master\examples\Boards_WiFi\ESP8266_Standalone\ESP8266_Standalone.ino:29:25: fatal error: ESP8266WiFi.h: No such file or directory

 #include <ESP8266WiFi.h>

                         ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.**

Anyone please guide me.
[image]

facchinm, 2016-06-27:
Hi @ashokelumalai, you are trying to compile for an Arduino UNO, while the sketch is designed for an ESP8266 board (https://github.com/blynkkk/blynk-library/blob/master/examples/Boards_WiFi/ESP8266_Standalone/ESP8266_Standalone.ino#L15). Simply follow the instructions on the sketch itself to get it working.

closed 2016-06-27 by facchinm

Flylingi, 2019-01-29:
Sorry for asking, but I have the same problem. Where can i find the instuctions to get it working?

per1234, 2019-01-29:
@Flylingi the problem has already been explained by facchinm.

This issue tracker is only to be used to report bugs or feature requests, not to ask for help with your project. The appropriate place to do that is the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to provide all the information we will need to help you.