arduino/arduino-builder #168: Parallelize builds

xcvista, 2016-07-09:
I seriously doubt if anyone here are still stuck with a single-core processor. My three computers have 2, 4 and 16 hyper-threading cores respectively. Since almost everyone have a multi-core (or at least hyper-threading) processor why not take advantage of it, by parallelizing the build process?

Multiple instances of GCC can be launched at the same time. So you can perform core count detection upon launch (people can upgrade their processors after installing the software) and when compiling, use as many cores as possible. Building multi-compile-unit code is an embarrassingly parallel problem.

per1234, 2016-07-10:
See https://github.com/arduino/arduino-builder/issues/17. It looks like some work has been done towards this in the [parallel branch](https://github.com/arduino/arduino-builder/tree/parallel).

cmaglie, 2016-08-26:
Duplicate of #17

closed 2016-08-26 by cmaglie