arduino/arduino-builder #170: Incorrect prototype when using constexpr

Chris--A, 2016-07-25:
Using 1.6.10 IDE compiled a few days ago.

This code requires a manual prototype, otherwise the IDE combines the `constexpr` and `int` causing an undeclared type error.

```C++
constexpr int test();
constexpr int test() { return 5; }

void setup() {
  Serial.begin(9600);
  Serial.println(test());
}

void loop() {
}
```

esorton, 2016-08-01:
I can confirm this issue on Windows 10/Arduino IDE 1.6.9 and 1.6.10.

matthijskooijman, 2016-08-01:
I can reproduce this as well. This seems to be a problem in ctags, which gets the returntype wrong.

    test    /tmp/arduino_modified_sketch_261104/sketch_aug01a.ino   /^constexpr int test() { return 5; }$/;"        kind:function   line:2  signature:()    returntype:constexprint

@igrr, perhaps we could persuade you to have a look at this? I would presume this is as simple as letting ctags know that `constexpr` is a keyword?

esorton, 2016-08-02:
Adding `constexpr` as a keyword in arduino/arduino-ctags fixed the issue for me.  Created pull request arduino/ctags#11.

matthijskooijman, 2016-08-02:
Awesome, thanks!

facchinm, 2016-12-19:
Fixed by merging https://github.com/arduino/ctags/pull/11

closed 2016-12-19 by facchinm