arduino/arduino-builder #176: SAML21 CMSIS Files

spiderkeys, 2016-08-03:
Would it be possible to update the CMSIS/Devices/ATMEL files to include the latest headers for the SAML21 CMSIS implementation? I'm developing a board core for the SAML21 series and noticed that while there is a saml.h, the include files it references are not present. It seems the folders are there for most of the other chips, so I was wondering if these could be included.

As a side note: Who is responsible for assembling all of the files inside the Devices/ATMEL folder in the tools? I know that Atmel put together the CMSIS implementations for each chip, but is it the Arduino team or Atmel that puts together the entire collection and writes the sam.h and sam<series>.h headers? Curious because I've found the individual chip folders in Atmel's ASF folders, but didn't know if there was an Atmel-provided collection of all of them as is present in the Arduino folder.

If Arduino maintains them and there is no specific process for updating it, I can add these in myself and submit a PR, if that would be helpful. Otherwise, it may be better to let the responsible party pull in the update from the official source.

ptibubu, 2016-12-16:
+1. Would be very interested if saml could be supported !

sandeepmistry, 2016-12-16:
@spiderkeys @ptibubu which core (SAMD, SAM) do you want you to use the CMSIS files with?

For the latest SAMD core release we've updated to use the following:

* https://github.com/arduino/ArduinoModule-CMSIS
* https://github.com/arduino/ArduinoModule-CMSIS-Atmel

ptibubu, 2016-12-16:
@sandeepmistry SAML21, which does not seem to be supported in CMSIS-Atmel.
See ASF 3.33 xdk-asf-3.33.0\sam0\utils\cmsis\saml21
Sorry if this is not the right place for this request, but having the same arduino package for all SAML21 core would be great.

sandeepmistry, 2016-12-23:
@aethaniel any ideas on why the SAML21 files are not in the [ArduinoModule-CMSIS-Atmel](https://github.com/arduino/ArduinoModule-CMSIS-Atmel)?

@ptibubu another option for you and @spiderkeys is to create a new package for your custom core or bundle them with the core. Is there are reason for not doing this?

aethaniel, 2016-12-23:
Hi Sandeep,

I can add any device on demand. I may have screened the content at some point but don't remember exactly.

I may propose a patch next week for saml21. Would that be ok for you?

Cheers,

Thibaut

sandeepmistry, 2016-12-23:
Hi Thibaut,

Sure, that would be great! We can ask @ptibubu  and @spiderkeys if it meets there needs.

Thanks!

spiderkeys, 2016-12-25:
Hi @sandeepmistry 

We ended up deciding not to use the SAML21 in our product some time back, so this is no longer an issue for me specifically. I did make an effort to gather up the SAML21 CMSIS files and bundle them in a way similar to Arduino, which you can find here:

https://github.com/OpenROV/saml21-experiment/tree/master/include/cmsis/atmel

These files were the latest versions included in Atmel Studio 7.0 as of 4 months ago. They correspond to what Atmel considers ASF 4.x, which is the version of ASF that their Atmel Start generator uses.

Note: While the SAML21 SERCOM peripherals operate in the exact same manner as the SAMD21, I believe that when I last worked on this, some of the macros in the component header files are missing in the SAML21 implementation that exist in the SAMD21 implementation. While I cannot remember the particulars, it seemed some of the I2CM macros were missing that end up getting used in Arduino's SERCOM class. Nothing that can't be easily patched in, but just a heads up if anyone decides to continue working on a core for this board.

ptibubu, 2016-12-30:
@sandeepmistry @aethaniel It will meet my need, thanks!

bluespider42, 2017-01-23:
I would also be very interested in a SAML21 Arduino core

aethaniel, 2017-01-23:
Hi Sandeep, saml21 wasn't expected to be integrated into any Arduino core afaik, the consistency being built around samd21 and samw25.

Furthermore Arduino core doesn't provide any low power features and api and therefore makes meaningless this introduction.

Well, i didn't follow closely the core api since some months and i may be wrong.

Cheers,
T.

sandeepmistry, 2017-01-24:
His @aethaniel,

There are no plans for SAML21 support in an official Arduino core. The original request is for the Atmel CMSIS package to include files for the SAML21 so that members from the community can build and maintain their own core.

aethaniel, 2017-01-24:
Got it, patch in progress on my side. Should be submitted by tomorrow.

aethaniel, 2017-01-25:
Patch submitted for test if someone want to give it a try.
https://github.com/arduino/ArduinoModule-CMSIS-Atmel/pull/3

closed 2017-03-17 by facchinm