arduino/arduino-builder #177: ctags not found on OSX

neilpa, 2016-08-29:
```sh
> D=/Applications/Arduino.app/Contents/Java
> $D/arduino-builder -hardware $D/hardware -tools $D/hardware/tools/avr -fqbn arduino:avr:uno test.ino
Build options changed, rebuilding all
fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory
```

Is there a way to disable the ctags usage or pass it the path that exists in the app package? I don't see anything in the help output and adding an extra `-tools $D/tools-builder/ctags/5.8-arduino10` to the above command didn't fix it either.

neilpa, 2016-08-29:
Looks like `-build-options-file` might do the trick. It's not very clear from the docs but poking around the source I suspect this is the format I need

```json
{ 
  "tools": {
    "ctags": {
      "path" : "/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino10"
    }
  }
}
```

neilpa, 2016-08-29:
Ok, so apparently I was way off the mark. It looks like the `build-options-file` is more like a [java *.properties file of the form `key=value`](https://github.com/arduino/arduino-builder/blob/a07ea16effee9c7402f877c063ac1ac4702b856d/src/arduino.cc/properties/properties.go#L105-L110). Might I recommend updating the [README to not call it a JSON](https://github.com/arduino/arduino-builder/tree/a07ea16effee9c7402f877c063ac1ac4702b856d#what-is-and-how-to-use-buildoptionsjson-file) file because this is rather confusing.

ricardojlrufino, 2016-10-26:
I got the same error using Linux 64 (arduino-builder-linux64-1.3.21)

> "{runtime.tools.ctags.path}/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino-sketch-8EEF6D2254B297BD93AC671AA24124F8/preproc/ctags_target_for_gcc_minus_e.cpp"
> fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory

ricardojlrufino, 2016-10-26:
Looking into other issues found the solution:
Change file: arduino-1.X.X/**hardware/platform.txt** , and change **tools.ctags.path**  to path of ctags, like:

> tools.ctags.path=/media/ricardo/arduino-1.6.9/tools-builder/ctags/5.8-arduino10

NicoHood, 2016-10-26:
Is there also a way to use the os ctags? that never worked for me.

ricardojlrufino, 2016-10-26:
I dont know

oqibidipo, 2016-10-28:
`man ctags` (OSX 10.11.6)
```
NAME
     ctags -- create a tags file

SYNOPSIS
     ctags [-BFadtuwvx] [-f tags_file] name ...

DESCRIPTION
     ctags makes a tags file for ex(1) from the specified C, Pascal, Fortran, YACC, lex, 
     and lisp sources. ...
```
Notice the absence of C++ as a supported language, so not very useful for arduino-builder.

NicoHood, 2016-10-28:
```
ctags --list-languages
Ant
Asm
Asp
Awk
Basic
BETA
C
C++
C#
Cobol
DosBatch
Eiffel
Erlang
Flex
Fortran
HTML
Java
JavaScript
Lisp
Lua
Make
MatLab
OCaml
Pascal
Perl
PHP
Python
REXX
Ruby
Scheme
Sh
SLang
SML
SQL
Tcl
Tex
Vera
Verilog
VHDL
Vim
YACC
```

facchinm, 2016-11-02:
@NicoHood , using the original `ctags` is not viable (the patchset over `master` is huge). `universal-ctags` is probably the way to go but needs a patched `arduino-builder` (the output syntax and the flags are not completely compatible)

closed 2017-03-17 by facchinm