arduino/arduino-builder #179: Um, ok, sparse instructions

treii28, 2016-09-02:
What is it supposed to do? I think I've gotten all installed through trial and error, I try to do what it says on the tests and nothing seems to happen:

go test arduino.cc/... -run 'TestPrototypesAdder.*'
package arduino.cc/arduino-builder
	imports runtime: C source files not allowed when not using cgo or SWIG: defs.c float.c heapdump.c lfstack.c malloc.c mcache.c mcentral.c mem_linux.c mfixalloc.c mgc0.c mheap.c msize.c os_linux.c os_linux_arm.c panic.c parfor.c proc.c runtime.c signal.c signal_arm.c signal_unix.c softfloat_arm.c stack.c string.c sys_arm.c vlrt.c

matthijskooijman, 2016-09-02:
Did you set $GOPATH (e.g. `export GOPATH=$(pwd)` from the root directory)? What are you showing in your post? Is that the output of the go test command? If not, what error do you get? I'm a bit puzzled as to what your asking really. I just tried your command in a git master checkout:

```
arduino-builder$ export GOPATH=$(pwd)
arduino-builder$ go test arduino.cc/... -run 'TestPrototypesAdder.*'                                                
?       arduino.cc/arduino-builder      [no test files]
?       arduino.cc/builder      [no test files]
?       arduino.cc/builder/builder_utils        [no test files]
?       arduino.cc/builder/constants    [no test files]
?       arduino.cc/builder/ctags        [no test files]
?       arduino.cc/builder/gohasissues  [no test files]
?       arduino.cc/builder/i18n [no test files]
?       arduino.cc/builder/phases       [no test files]
ok      arduino.cc/builder/test 3.892s
?       arduino.cc/builder/types        [no test files]
?       arduino.cc/builder/utils        [no test files]
ok      arduino.cc/properties   0.004s
ok      arduino.cc/timeutils    0.004s
```

matthijskooijman, 2016-12-21:
I'm closing this issue for lack of reply.

closed 2016-12-21 by matthijskooijman