arduino/arduino-builder #180: Auto generated function declarations no longer works for callbacks assigned at variable declaration

AlexanderBrevig, 2016-09-28:
Here's and example that will break the auto generation of function declarations in the IDE, not sure for how long but I've gotten regular support request for a good while now.
The only thing I know, this used to work.

```
// say you have a library that accepts a callback
class BUG {public:BUG(void (*fn)(void)){}};

//void makeItBreak(){} //uncomment to fix
BUG b(makeItBreak); //this will break

void setup(){}
void loop(){}
void makeItBreak(){}```

facchinm, 2016-12-06:
Hi @AlexanderBrevig,
#191 contains a fix for this, however I'm not entirely happy about that patch (too much guesswork). If you change your code in the (semantic equivalent) 
```C++
class BUG {public:BUG(void (*fn)(void)){}};

BUG b(&makeItBreak); //this will break

void setup(){}
void loop(){}
void makeItBreak(){}
```
it works correctly with released builder.
Do you think it can be an acceptable solution?

AlexanderBrevig, 2016-12-08:
It is an improvement but does not help in that it still is not 'backward compatible'. If I have to choose explaining moving a function above it, and adding the & I'll go for the former.

I'd personally still like this to be included though :) personally it's a win! Sadly though, not for the target audience.

facchinm, 2017-09-12:
Fixed by 3282535

closed 2017-09-12 by facchinm