arduino/arduino-builder #185: preprocessor fails on #include "ESP8266WiFiMulti.h" (for external ESP8266 addon)

19eighties, 2016-10-10:
Reference: https://github.com/arduino/arduino-builder/issues/68

reproduce by:

add http://arduino.esp8266.com/stable/package_esp8266com_index.json to board manager prefs
use board manager to install ESP8266 support
try to compile this sketch
```
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
void setup() {}
void loop() { hello(); }
void hello(void) {}

```
Error output:
`hunt-compile-error:4: error: 'hello' was not declared in this scope
 void loop() { hello(); }
                     ^
exit status 1
'hello' was not declared in this scope`

Using 1.6.9 on Mac OS 10.11.16.

facchinm, 2016-10-11:
This was solved between IDE 1.6.9 and 1.6.10. Your sketch compiles correctly now (latest IDE is 1.6.12), so I'm closing this as solved

closed 2016-10-11 by facchinm

19eighties, 2016-10-11:
OK, I didn't realize I was that far behind, thanks.