arduino/arduino-builder #186: exec: "{runtime.tools.ctags.path}/ctags": file does not exist

Caizhenyu, 2016-10-28:
I am a developer for arduino ide now.I use this arduino ide and add my funcation.Now the ide can not compile and throw that,"exec: "{runtime.tools.ctags.path}/ctags": file does not exist".
I hava seen other issuse the same ,but I can not resolve it.
Would anyone can help me ?

Caizhenyu, 2016-10-28:
That is in Windows platform.

facchinm, 2016-10-28:
Hi @Caizhenyu ,
can you post the complete specifications of your installation (Windows version, IDE version, installed cores, verbose compilation output) ? Thanks

Caizhenyu, 2016-10-28:
Windows 10, Arduino 1.6.11, this all i know about installation.

facchinm, 2016-10-28:
Could you test latest IDE (1.6.12) or nightly version? The get the verbose output, select "Show verbose output during compilation" in IDE File/Preferences, run a compile of `Blink` example and post the output here

Caizhenyu, 2016-10-28:
code  
```
setup{}
loop{}
```

console
```
E:\arduino\arduino-builder -dump-prefs -logger=machine -hardware E:\arduino\hardware -tools E:\arduino\hardware\tools\avr -built-in-libraries E:\arduino\libraries -libraries E:\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10612 -build-path C:\Users\myname\AppData\Local\Temp\arduino_build_939366 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=E:\arduino\hardware\tools\avr -verbose C:\Users\myname\Desktop\sketch_oct17a\sketch_oct17a.ino
E:\arduino\arduino-builder -compile -logger=machine -hardware E:\arduino\hardware -tools E:\arduino\hardware\tools\avr -built-in-libraries E:\arduino\libraries -libraries E:\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10612 -build-path C:\Users\myname\AppData\Local\Temp\arduino_build_939366 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=E:\arduino\hardware\tools\avr -verbose C:\Users\myname\Desktop\sketch_oct17a\sketch_oct17a.ino
Using board 'uno' from platform in folder: E:\arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: E:\arduino\hardware\arduino\avr
Detecting libraries used...
"E:\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "C:\Users\myname\AppData\Local\Temp\arduino_build_939366\sketch\sketch_oct17a.ino.cpp" -o "nul"
"E:\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "C:\Users\myname\AppData\Local\Temp\arduino_build_939366\sketch\sketch_oct17a.ino.cpp" -o "nul"
Generating function prototypes...
"E:\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "C:\Users\myname\AppData\Local\Temp\arduino_build_939366\sketch\sketch_oct17a.ino.cpp" -o "C:\Users\myname\AppData\Local\Temp\arduino_build_939366\preproc\ctags_target_for_gcc_minus_e.cpp"
"{runtime.tools.ctags.path}/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\myname\AppData\Local\Temp\arduino_build_939366\preproc\ctags_target_for_gcc_minus_e.cpp"
exec: "{runtime.tools.ctags.path}/ctags": file does not exist
for BoardPort Arduino/Genuino Uno complile error.
```

facchinm, 2016-10-28:
Ok, the commandline is missing `-tools E:\arduino\tools-builder`. Can you check if that directory exists and contains `ctags` ?

Caizhenyu, 2016-10-29:
thank you.
i add the tools-builder and ctags ,it throw a new error. is my arduino-core missing some floder?

code:
```
setup[]
loop{}
```
console:
```
E:\arduino\arduino-builder -dump-prefs -logger=machine -hardware E:\arduino\hardware -tools E:\arduino\tools-builder -tools E:\arduino\hardware\tools\avr -built-in-libraries E:\arduino\libraries -libraries E:\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10612 -build-path C:\Users\myname\AppData\Local\Temp\arduino_build_562314 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=E:\arduino\hardware\tools\avr -verbose C:\Users\myname\Desktop\sketch_oct17a\sketch_oct17a.ino
E:\arduino\arduino-builder -compile -logger=machine -hardware E:\arduino\hardware -tools E:\arduino\tools-builder -tools E:\arduino\hardware\tools\avr -built-in-libraries E:\arduino\libraries -libraries E:\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10612 -build-path C:\Users\myname\AppData\Local\Temp\arduino_build_562314 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=E:\arduino\hardware\tools\avr -verbose C:\Users\myname\Desktop\sketch_oct17a\sketch_oct17a.ino
Using board 'uno' from platform in folder: E:\arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: E:\arduino\hardware\arduino\avr
Detecting libraries used...
"E:\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\sketch\sketch_oct17a.ino.cpp" -o "nul"
"E:\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\sketch\sketch_oct17a.ino.cpp" -o "nul"
Generating function prototypes...
"E:\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\sketch\sketch_oct17a.ino.cpp" -o "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\preproc\ctags_target_for_gcc_minus_e.cpp"
"E:\arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\preproc\ctags_target_for_gcc_minus_e.cpp"
compling...
"E:\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\sketch\sketch_oct17a.ino.cpp" -o "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\sketch\sketch_oct17a.ino.cpp.o"
Compiling libraries...
Compiling core...
"E:\arduino\hardware\tools\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "E:\arduino\hardware\arduino\avr\cores\arduino\wiring_pulse.S" -o "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\core\wiring_pulse.S.o"
"E:\arduino\hardware\tools\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\arduino\hardware\arduino\avr\cores\arduino" "-IE:\arduino\hardware\arduino\avr\variants\standard" "E:\arduino\hardware\arduino\avr\cores\arduino\WInterrupts.c" -o "C:\Users\myname\AppData\Local\Temp\arduino_build_562314\core\WInterrupts.c.o"
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.

exit status 1
for BoardPort Arduino/Genuino Uno complie error.
```

facchinm, 2016-11-02:
Hi @Caizhenyu ,
you shouldn't need to add `ctags` since it is shipped with the IDE. Also #188 is caused by very very strange configuration problems. I'd ask you to completely clean up the IDE and Arduino15 folder.
* Uninstall the IDE via Control Panel
* Remove folder `E:/Arduino`
* Remove folder `$USER\AppData\Local\Arduino15` (if it exists) -> $USER is your "home" folder
* Remove folder `$USER\AppData\Roaming\Arduino15` (if it exists)

Then reinstall IDE 1.6.12 using the installer. In this way, no stray old avr core should be picked up and all the tools should be in the right place.

Caizhenyu, 2016-11-03:
i build arduino ide along the case [https://github.com/arduino/Arduino/wiki/Building-Arduino-IDE](url) ,but the ide exit while starting,i see the start interface and then it exit.
[image]
[image]

closed 2017-03-17 by facchinm

Marti2203, 2020-10-04:
Hi, I am just having this issue on a Linux Open Suse machine. I know it is a closed comment but I would like to get some help if possible

cyc1ingsir, 2020-11-02:
Regarding opensuse: I had the same problem but have found a solution:
`fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory`

I do want to use the IDE to put something on a ESP8266 just to put it into context but with a little variation it might be work for other platforms as well:
The Arduino IDE installed is from this source:
```
zypper info Arduino
Repository     : arduino
Name           : Arduino
Version        : 1.8.13-1.4
Arch           : x86_64
Vendor         : obs://build.opensuse.org/CrossToolchain:avr
```
Since I have ctags intalled from:
```
Repository     : arduino
Name           : ctags
Version        : 5.8-47.1
Arch           : x86_64
Vendor         : obs://build.opensuse.org/CrossToolchain:avr
```
this can be used:
```
which ctags
/usr/bin/ctags
ctags --version
Exuberant Ctags Development, Copyright (C) 1996-2009 Darren Hiebert
  Addresses: <dhiebert@users.sourceforge.net>, http://ctags.sourceforge.net
  Optional compiled features: +wildcards, +regex
```
The ESP8266 boards option was installed via the board manager which installed into `~/.arduino15/packages/esp8266/hardware/esp8266/2.7.4` including a `platform.txt` in this folder.
To this file I appended a line:
```
echo "tools.ctags.path=/usr/bin/" >> platform.txt
```
:tada: 
Just for completeness, this are the avr related packages on my tumbleweed installation:
```
rpm -qa | grep avr
cross-avr-gcc8-8.4.0+git2127-2.75.x86_64
libavresample3-3.4.5-4.42.x86_64
liblavrec-2_0-0-2.0.0-27.9.x86_64
libavresample2-2.8.15-8.2.x86_64
cross-avr-binutils-2.35-398.2.x86_64
libavresample4_0-4.3.1-8.1.x86_64
avrdude-6.3-8.48.x86_64
avr-libc-2.0.0-14.385.noarch
```
I hope this will be helpful for someone.

Marti2203, 2020-11-02:
My solution was that I added some extra files to the Arduino folder - making a folder called tools and in tools-builder I added a folder called ctags with a symlink to ctags and a tool folder with another symlink. I am pretty sure you need only tools/ctag/tool/ctags but I did not play around with it a lot after making it work as it was for a small project.

Pippadi, 2020-11-29:
I had problems with the Arduino IDE, openSUSE Tumbleweed, and ESP8266 as well.
`zypper info Arduino`:
```
Repository     : Crosscompiling Toolchain for ATMEL AVR microcontrollers (openSUSE_Tumbleweed)
Name           : Arduino
Version        : 1.8.13-1.5
Arch           : x86_64
Vendor         : obs://build.opensuse.org/CrossToolchain:avr
```

The complaints:
```processing.app.debug.RunnerException
	at cc.arduino.Compiler.callArduinoBuilder(Compiler.java:317)
	at cc.arduino.Compiler.loadPreferences(Compiler.java:216)
	at cc.arduino.Compiler.build(Compiler.java:175)
	at processing.app.SketchController.build(SketchController.java:664)
	at processing.app.Editor$BuildHandler.run(Editor.java:1632)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.io.IOException: Cannot run program "/usr/lib64/Arduino-1.8.13/arduino-builder": error=2, No such file or directory
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
	at java.base/java.lang.Runtime.exec(Runtime.java:592)
	at java.base/java.lang.Runtime.exec(Runtime.java:451)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.Compiler.callArduinoBuilder(Compiler.java:297)
	... 5 more
Caused by: java.io.IOException: error=2, No such file or directory
	at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
	at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:340)
	at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:271)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
	... 10 more
```

I got rid of this by making a symlink to the file it wasn't able to find:
`sudo ln -s /usr/bin/arduino-builder /usr/lib64/Arduino-1.8.13/arduino-builder`

Then, I got the same `fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory` error.

@cyc1ingsir's solution worked for me too (Thanks!):
`echo "tools.ctags.path=/usr/bin/" >> .arduino15/packages/esp8266/hardware/esp8266/2.7.4/platform.txt`

I have no idea of the intricacies of the Arduino IDE, and whether there's a better way to do this, but this worked. Hope it helps someone.

matthijskooijman, 2020-11-30:
> I had problems with the Arduino IDE, openSUSE Tumbleweed, and ESP8266 as well.

Sounds like you're installing the OpenSUSE packaged version of Arduino? Then we don't know the intricacies of how that packaging was done (the official Arduino IDE zipfile bundles things like java and gcc, distribution packages have a tendency to split things into different packages in ways that might cause problems like these). So, for your case, I would suggest reporting a bug with OpenSUSE, I suspect this is something that needs to be fixed on their end (to double-check, you can download and unpack the zip from arduino.cc, no install required, and see if that *does* work without the workaround. If that works, likely an OpenSUSE bug that should be reported).

Pippadi, 2020-11-30:
I installed from [here](https://build.opensuse.org/project/show/CrossToolchain:avr), which seems to be the most current source for Arduino in OpenSUSE's repositories.
I just downloaded the arduino.cc zip, and sure enough, it worked without my workarounds. After some digging, I found that this bug was already reported in OpenSUSE.
Thanks for the clarification!

asdil12, 2022-01-10:
@Pippadi could you post the bugzilla link?

Pippadi, 2022-01-10:
@asdil12 This one: [https://bugzilla.opensuse.org/show_bug.cgi?id=1178153](https://bugzilla.opensuse.org/show_bug.cgi?id=1178153)
Seems to have been fixed a while back. Everything works properly for me now.

IPlayZed, 2022-04-15:
> I had problems with the Arduino IDE, openSUSE Tumbleweed, and ESP8266 as well. `zypper info Arduino`:
>
> ```
> Repository     : Crosscompiling Toolchain for ATMEL AVR microcontrollers [...]

I a on OpenSUSE Thumbleweed and installed Arduino through Zypper ad the OpenSUSE Wiki suggests. This will fix the isse, can confirm.

Pippadi, 2022-04-19:
I continue to get the `fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory` error when compiling for ESP32 as well, _from the Arduino IDE_. Using `arduino-cli compile` works fine, and subsequent compiles from the IDE and `arduino --verify` now work fine as well, even without adding that line to `platform.txt`.

If I had to speculate, I'd say that it isn't a packaging issue, but it isn't really a board-specific issue either, considering that it seems to work fine with other Linuxes.

per1234, 2022-04-19:
>Using arduino-cli compile works fine, and subsequent compiles from the IDE and arduino --verify now work fine as well, even without adding that line to platform.txt.

This is because, unlike Arduino IDE 1.x, Arduino CLI automatically installs ctags from Arduino's download server if it is not already present (by default, it is installed under `~/.arduino15/packages/builtin/tools/ctags`. The reason for the different behaviors is that Arduino IDE 1.x is designed to come with a bundled installation of ctags. So Arduino IDE 1.x won't install ctags, but it will use the copy that was installed by Arduino CLI if present.

It is common practice for the 3rd party package creators to remove the bundled components from the Arduino IDE installation, since the package repository maintainers don't approve of it.

MohammedNabilFarag, 2024-12-12:
exec: "{runtime.tools.ctags.path}/ctags": executable file not found in %PATH%

Compilation error: exec: "{runtime.tools.ctags.path}/ctags": executable file not found in %PATH%

**Note that the Arduino version is (arduino-ide_2.3.3_Windows_64bit)**