arduino/arduino-builder #187: arduino-builder not installed by default with OSX?

jdar, 2016-10-30:
Is this something I can only get if I compile the arduino project manually?

I installed using homebrew. Next I was expecting to find an executable, and add it to $PATH, but there is no 'arduino' executable:

$ find  /Applications/Arduino.app/ -name "arduino*" -type f
/Applications/Arduino.app//Contents/Java/arduino-core.jar
/Applications/Arduino.app//Contents/Java/hardware/arduino/avr/firmwares/wifishield/wifi_dnld/src/SOFTWARE_FRAMEWORK/BOARDS/ARDUINO/arduino.h
/Applications/Arduino.app//Contents/Java/hardware/arduino/avr/firmwares/wifishield/wifiHD/src/SOFTWARE_FRAMEWORK/BOARDS/ARDUINO/arduino.h
/Applications/Arduino.app//Contents/Java/lib/arduino_icon.ico
/Applications/Arduino.app//Contents/Java/libraries/TFT/examples/Arduino/TFTBitmapLogo/arduino.bmp
/Applications/Arduino.app//Contents/Java/libraries/TFT/examples/Esplora/EsploraTFTBitmapLogo/arduino.bmp
/Applications/Arduino.app//Contents/Java/libraries/WiFi/extras/wifi_dnld/src/SOFTWARE_FRAMEWORK/BOARDS/ARDUINO/arduino.h
/Applications/Arduino.app//Contents/Java/libraries/WiFi/extras/wifiHD/src/SOFTWARE_FRAMEWORK/BOARDS/ARDUINO/arduino.h
/Applications/Arduino.app//Contents/Java/reference/arduino.cc/css/arduino-icons.css
/Applications/Arduino.app//Contents/Java/reference/arduino.cc/css/arduino_code_highlight.css
/Applications/Arduino.app//Contents/Java/reference/arduino.cc/en/pub/skins/arduinoWide_SSO/css/arduinoWide_SSO.css
/Applications/Arduino.app//Contents/Java/reference/arduino.cc/en/Reference/arduino.cc
/Applications/Arduino.app//Contents/Java/reference/arduino.cc/en/uploads/Reference/arduino_mega_ethernet_pins.png
/Applications/Arduino.app//Contents/Java/reference/arduino.cc/en/uploads/Reference/arduino_uno_ethernet_pins.png
/Applications/Arduino.app//Contents/Java/reference/arduino.cc/en/uploads/Reference/arduinoEthernetShieldSD.jpg
/Applications/Arduino.app//Contents/Java/reference/Edison_help_files/arduinoUno.css
/Applications/Arduino.app//Contents/Java/reference/Edison_help_files/arduinoWide.css
/Applications/Arduino.app//Contents/Java/reference/Edison_help_files/arduinoWideRender.css
/Applications/Arduino.app//Contents/Java/reference/Galileo_help_files/arduinoUno.css
/Applications/Arduino.app//Contents/Java/reference/Galileo_help_files/arduinoWide.css
/Applications/Arduino.app//Contents/Java/reference/Galileo_help_files/arduinoWideRender.css

jdar, 2016-10-30:
I'm on 10.11.6 El Capitan, if it matters.

facchinm, 2016-11-02:
Hi @jdar, the executable name is `Arduino` (with capital A) and you can find it in `Arduino.app/Contents/MacOS`. Hope it helps!

facchinm, 2016-11-02:
`arduino-builder`, instead, can be found in `Arduino.app//Contents/Java/` if you installed a fairly new ( > 1.6.6) version of the IDE

closed 2016-11-02 by facchinm