arduino/arduino-builder #197: Prototypes are not generated in multi-file sketches

me21, 2016-12-09:
Looks similar to #68 and #185, but on Arduino 1.6.12.

reproduce by:

* create sketch main.ino
```
void setup() {
  // put your setup code here, to run once:
  sub();
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

* create sketch sub.ino
```
void sub() {
  
}
```

> main:3: error: 'sub' was not declared in this scope

Tried that on esp8266 and arduino cores. Arduino IDE v1.6.12.

me21, 2016-12-09:
Seems that Arduino does not like non-latin letters in path to sketches folder. Cyrillic in my case. Changing sketch folder path to the one which does not contain cyrillic letters fixed this for me.

Still, I'm keeping this issue open because it's not good to require latin-only paths.

facchinm, 2016-12-12:
The issue has been solved by https://github.com/arduino/arduino-builder/commit/46410677e92f51aca90fc9a4689f08eee766ff5f , so next IDE version will contain the proper fix. If you want to test the fix, download the latest release from https://github.com/arduino/arduino-builder/releases/tag/1.3.22 and replace both `arduino-builder` and `ctags` with the downloaded ones

closed 2016-12-12 by facchinm