arduino/arduino-builder #199: Prototype generation fails if primary file is empty

facchinm, 2016-12-19:
_Moved from https://github.com/arduino/Arduino/issues/5737 by @alto777_

The attached sketch will not compile with IDE 1.6.13.
[apUNI8a.zip](https://github.com/arduino/arduino-builder/files/660586/apUNI8a.zip)

If two blank lines are removed from apUNI8a.ino as stated in the comment preceding them, the sketch will compile.

The matter is discussed and concluded in

http://forum.arduino.cc/index.php?topic=442502.15

facchinm, 2016-12-19:
If the two blank lines are not removed, the prototypes are being inserted inside the comment section
```C
#include <Arduino.h>
#line 1 "/tmp/apUNI8a/apUNI8a.ino"
#line 1 "/tmp/apUNI8a/apUNI8a.ino"
// This file makes the Arduino IDE happy (directory name must match a *.ino file).
/* add a comment */
// Code is in all other files
// this did bind to the Hubsan X4 H107L.
// REMOVE THE TWO BLANK LINES AFTER THIS LINE and it compiles. 1.6.13

// see myA7XShouting.txt

// is there such a thing as overrun?
// HARD CODED channel 0x50 everywhere
// THIS DOES single packets between two instances!

// now add - PB0 asserted on reset means listen always and ack after a delay any received meesages
// without discrimination, this should set up a pinging pong.

// starting to strip out some of the Hubsan stuff...

/* add a comment */
/*
was successful ping-pong unit program
now this section will model the hand-rolled ack pack
then both will try it, here's where I'll add the prime table.

milliseconds?
#line 26 "/tmp/apUNI8a/a7105.ino"
void A7105_Setup();
#line 43 "/tmp/apUNI8a/a7105.ino"
void A7105_Reset();
#line 52 "/tmp/apUNI8a/a7105.ino"
int A7105_calibrate_IF();
#line 87 "/tmp/apUNI8a/a7105.ino"
int A7105_calibrate_VCB(uint8_t channel);
#line 120 "/tmp/apUNI8a/a7105.ino"
void A7105_SetPower(int power);
#line 190 "/tmp/apUNI8a/a7105.ino"
void A7105_WriteReg(uint8_t address, uint8_t data);
#line 197 "/tmp/apUNI8a/a7105.ino"
uint8_t A7105_ReadReg(uint8_t address);
#line 211 "/tmp/apUNI8a/a7105.ino"
void A7105_WriteData(uint8_t *dpbuffer, uint8_t len, uint8_t channel);
#line 229 "/tmp/apUNI8a/a7105.ino"
void A7105_ReadData(uint8_t *dpbuffer, uint8_t len);
#line 247 "/tmp/apUNI8a/a7105.ino"
void A7105_WriteID(unsigned long id);
#line 258 "/tmp/apUNI8a/a7105.ino"
void A7105_ReadID();
#line 275 "/tmp/apUNI8a/a7105.ino"
void make_test_packet(uint8_t testpacket[]);
#line 295 "/tmp/apUNI8a/a7105.ino"
void printpacket(uint8_t packet[]);
#line 308 "/tmp/apUNI8a/a7105.ino"
void A7105_shoutchannel();
#line 337 "/tmp/apUNI8a/a7105.ino"
int A7105_oneShout();
#line 361 "/tmp/apUNI8a/a7105.ino"
int A7105_oneShoutRAM(unsigned char xState, unsigned char xType);
#line 391 "/tmp/apUNI8a/a7105.ino"
void eavesdrop();
#line 442 "/tmp/apUNI8a/a7105.ino"
uint8_t A7105_findchannel();
#line 464 "/tmp/apUNI8a/a7105.ino"
int A7105_sniffchannel();
#line 484 "/tmp/apUNI8a/a7105.ino"
void A7105_sniffchannel(uint8_t _channel);
#line 498 "/tmp/apUNI8a/a7105.ino"
void A7105_scanchannels(const uint8_t channels[]);
#line 8 "/tmp/apUNI8a/hubsan.ino"
static void initialize();
#line 22 "/tmp/apUNI8a/hubsan.ino"
int hubsan_init();
#line 80 "/tmp/apUNI8a/hubsan.ino"
static void update_crc();
#line 63 "/tmp/apUNI8a/main.ino"
void setup();
#line 116 "/tmp/apUNI8a/main.ino"
void loop();
#line 165 "/tmp/apUNI8a/main.ino"
void listeni();
#line 172 "/tmp/apUNI8a/main.ino"
void kickIt(unsigned char xType);
#line 181 "/tmp/apUNI8a/main.ino"
void lightFSM();
#line 188 "/tmp/apUNI8a/main.ino"
void touchFSM();
#line 207 "/tmp/apUNI8a/main.ino"
void fsmFSM();
#line 243 "/tmp/apUNI8a/main.ino"
void doBW(void);
#line 249 "/tmp/apUNI8a/main.ino"
void checkBW();
#line 260 "/tmp/apUNI8a/main.ino"
unsigned char getSwitches();
#line 294 "/tmp/apUNI8a/main.ino"
void setLEDs(unsigned char xx);
#line 312 "/tmp/apUNI8a/main.ino"
void blinkLEDs();
#line 11 "/tmp/apUNI8a/protocol.ino"
void PROTOCOL_SetBindState(uint32_t msec);
#line 26 "/tmp/apUNI8a/a7105.ino"
*/

/* add a comment */

#line 1 "/tmp/apUNI8a/a7105.ino"
/*
...
```
Trying to solve it right now, it won't probably make it into 1.6.14 but surely into next

bperrybap, 2016-12-19:
Also, just in case it matters, if you add a dummy line of C code to the bottom of the file, it works again as well.
i.e.
If you insert
```
void dummy(void) {}
```
at the bottom of that ino it works again.

commit 9b3f6b81 (2017-06-14): Only trust function line number in main sketch file

Previously, function line numbers are extracted from any tag (in any file) so exctracted line is nonsense if the main sketch is empty/has lots of lines and lots of comments/etc

Fixes #199
closed 2017-06-14 by facchinm