arduino/arduino-builder #202: Tests fail with self compiled ctags

NicoHood, 2017-01-06:
I am using my own arduino-ctags build in order to use the ArchLinux system shared libraries and apply a security patch. However it at compiling a wiring c file. With the builtin ctags this works as expected. Any ideas why?

The Arch PKGBUILD is here:
https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/arduino-ctags

I've applied this patch, but it seems to only be important for javascript:
https://git.archlinux.org/svntogit/community.git/tree/trunk/CVE-2014-7204.patch?h=packages/arduino-ctags

This is what I patch:
```batch

prepare() {
	cd "${srcdir}/${pkgname}-${pkgver}/src/arduino.cc/builder/hardware/"
	sed -i 's#^tools.ctags.path=.*#tools.ctags.path=/usr/bin#' platform.txt
	sed -i 's#^tools.ctags.cmd.path=.*#tools.ctags.cmd.path={path}/arduino-ctags#' platform.txt
}
```

This is how the tests start
```
==> Starting check()...
Running check() for the first time might take a while as it downloads ~1GB of testfiles. Skip with --nocheck
?   	arduino.cc/arduino-builder	[no test files]
?   	arduino.cc/builder	[no test files]
?   	arduino.cc/builder/builder_utils	[no test files]
?   	arduino.cc/builder/constants	[no test files]
ok  	arduino.cc/builder/ctags	0.004s
?   	arduino.cc/builder/gohasissues	[no test files]
?   	arduino.cc/builder/i18n	[no test files]
ok  	arduino.cc/builder/phases	0.003s
Downloading http://downloads.arduino.cc/cores/avr-1.6.10.tar.bz2
Downloading http://downloads.arduino.cc/cores/sam-1.6.7.tar.bz2
Downloading http://downloads.arduino.cc/cores/samd-1.6.5.tar.bz2
Downloading http://downloads.arduino.cc/tools/avrdude-6.0.1-arduino5-x86_64-pc-linux-gnu.tar.bz2
Downloading http://downloads.arduino.cc/tools/avr-gcc-4.8.1-arduino5-x86_64-pc-linux-gnu.tar.bz2
Downloading http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-linux64.tar.gz
Downloading http://downloads.arduino.cc/tools/ctags-5.8-arduino11-x86_64-pc-linux-gnu.tar.bz2
Downloading http://downloads.arduino.cc/bossac-1.6.1-arduino-x86_64-linux-gnu.tar.gz
Downloading http://downloads.arduino.cc/bossac-1.5-arduino2-x86_64-linux-gnu.tar.gz
Downloading http://downloads.arduino.cc/tools/OpenOCD-0.9.0-arduino-x86_64-linux-gnu.tar.bz2
Downloading http://downloads.arduino.cc/CMSIS-4.0.0.tar.bz2
Downloading http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-linux64.tar.gz
Downloading https://redbearlab.github.io/arduino/Blend/blend_boards.zip
Downloading http://downloads.arduino.cc/libraries/arduino-libraries/Audio-1.0.4.zip
Downloading http://downloads.arduino.cc/libraries/adafruit/Adafruit_PN532-1.0.0.zip
Downloading http://downloads.arduino.cc/libraries/arduino-libraries/Bridge-1.6.1.zip
Downloading http://downloads.arduino.cc/libraries/PaulStoffregen/CapacitiveSensor-0.5.0.zip
Downloading http://downloads.arduino.cc/libraries/arduino-libraries/Ethernet-1.1.1.zip
Downloading http://downloads.arduino.cc/libraries/arduino-libraries/Robot_IR_Remote-1.0.2.zip
Downloading http://downloads.arduino.cc/libraries/FastLED/FastLED-3.1.0.zip
Warning: Board my_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
Warning: Board my_symlinked_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
Warning: Board my_symlinked_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
Ts: 1483692419 - Running: GenerateBuildPathIfMissing
Ts: 1483692419 - Running: EnsureBuildPathExists
Ts: 1483692419 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1483692419 - Running: AddAdditionalEntriesToContext
Ts: 1483692419 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1483692419 - Running: HardwareLoader
Ts: 1483692419 - Running: PlatformKeysRewriteLoader
Ts: 1483692419 - Running: RewriteHardwareKeys
Ts: 1483692419 - Running: ToolsLoader
Ts: 1483692419 - Running: TargetBoardResolver
Using board 'uno' from platform in folder: /build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr
Using core 'arduino' from platform in folder: /build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr
Ts: 1483692419 - Running: AddBuildBoardPropertyIfMissing
Ts: 1483692419 - Running: LibrariesLoader
Ts: 1483692419 - Running: SketchLoader
Ts: 1483692419 - Running: SetupBuildProperties
Ts: 1483692419 - Running: LoadVIDPIDSpecificProperties
[... 194 lines cut ...]
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/hooks.c" -o "/tmp/test404130681/core/hooks.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/wiring.c" -o "/tmp/test404130681/core/wiring.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/wiring_analog.c" -o "/tmp/test404130681/core/wiring_analog.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/wiring_digital.c" -o "/tmp/test404130681/core/wiring_digital.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hard [... cut ...]
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-ar" rcs  "/tmp/test404130681/core/core.a" "/tmp/test404130681/core/WInterrupts.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-ar" rcs  "/tmp/test404130681/core/core.a" "/tmp/test404130681/core/hooks.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-ar" rcs  "/tmp/test404130681/core/core.a" "/tmp/test404130681/core/wiring.c.o"==> ERROR: Build failed, check /var/lib/archbuild/extra-x86_64/arch/build
```

facchinm, 2017-01-09:
Hi Nico,
`ctags` 5.8-arduino11 was released but the tag was not pushed, so a test was failing. Tag pushed and it should be ok now :wink:

closed 2017-01-09 by facchinm

NicoHood, 2017-01-09:
@facchinm thanks, i will test. Could you please sign the new release?

NicoHood, 2017-01-09:
It still fails:
```
Compiling sketch...
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/tmp/test985855614/sketch/sketch_with_multiline_prototypes.ino.cpp" -o "/tmp/test985855614/sketch/sketch_with_multiline_prototypes.ino.cpp.o"
Compiling libraries...
Compiling core...
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/test985855614/core/wiring_pulse.S.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/tmp/test985855614/core/WInterrupts.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/hooks.c" -o "/tmp/test985855614/core/hooks.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/wiring.c" -o "/tmp/test985855614/core/wiring.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino" "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/variants/leonardo" "/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_hardware/arduino/avr/cores/arduino/wiring_analog.c" -o "/tmp/test985855614/core/wiring_analog.c.o"
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino. [... cut ...]
"/build/arduino-builder/src/arduino-builder-1.3.23/src/arduino.cc/builder/test/downloaded_tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-objcopy" -O binary  "/tmp/test175014853/sketch12.ino.elf" "/tmp/test175014853/sketch12.ino.bin"
Sketch uses 8076 bytes (3%) of program storage space. Maximum is 262144 bytes.
Build options changed, rebuilding all
FAIL
FAIL	arduino.cc/builder/test	629.807s
?   	arduino.cc/builder/types	[no test files]
?   	arduino.cc/builder/utils	[no test files]
ok  	arduino.cc/properties	0.007s
ok  	arduino.cc/timeutils	0.012s
==> ERROR: A failure occurred in check().
    Aborting...
==> ERROR: Build failed, check /var/lib/archbuild/testing-x86_64/arch/build
```

facchinm, 2017-01-10:
There is something wrong going on here. I tested your last PKGBUILD (with the patches you provided) and the errors  get are all related to that patches
```bash
[martino@laptop arduino-builder]$ go test -v arduino.cc/... -run 'TestLoadHardwareMixingUserHardwareFolder'
?   	arduino.cc/arduino-builder	[no test files]
?   	arduino.cc/builder	[no test files]
?   	arduino.cc/builder/builder_utils	[no test files]
?   	arduino.cc/builder/constants	[no test files]
PASS
ok  	arduino.cc/builder/ctags	0.003s
?   	arduino.cc/builder/gohasissues	[no test files]
?   	arduino.cc/builder/i18n	[no test files]
PASS
ok  	arduino.cc/builder/phases	0.004s
=== RUN   TestLoadHardwareMixingUserHardwareFolder
--- FAIL: TestLoadHardwareMixingUserHardwareFolder (0.00s)
        Error Trace:    hardware_loader_test.go:146
	Error:		Not equal: "{runtime.tools.ctags.path}" (expected)
			        != "/usr/bin" (actual)
		
FAIL
FAIL	arduino.cc/builder/test	0.005s
?   	arduino.cc/builder/types	[no test files]
?   	arduino.cc/builder/utils	[no test files]
PASS
ok  	arduino.cc/properties	0.011s
PASS
ok  	arduino.cc/timeutils	0.007s
[martino@laptop arduino-builder]$ go test -v arduino.cc/... -run 'TestLoadHardware'
?   	arduino.cc/arduino-builder	[no test files]
?   	arduino.cc/builder	[no test files]
?   	arduino.cc/builder/builder_utils	[no test files]
?   	arduino.cc/builder/constants	[no test files]
PASS
ok  	arduino.cc/builder/ctags	0.004s
?   	arduino.cc/builder/gohasissues	[no test files]
?   	arduino.cc/builder/i18n	[no test files]
PASS
ok  	arduino.cc/builder/phases	0.025s
=== RUN   TestLoadHardware
--- FAIL: TestLoadHardware (0.00s)
        Error Trace:    hardware_loader_test.go:81
	Error:		Not equal: "{runtime.tools.ctags.path}" (expected)
			        != "/usr/bin" (actual)
		
=== RUN   TestLoadHardwareMixingUserHardwareFolder
--- FAIL: TestLoadHardwareMixingUserHardwareFolder (0.00s)
        Error Trace:    hardware_loader_test.go:146
	Error:		Not equal: "{runtime.tools.ctags.path}" (expected)
			        != "/usr/bin" (actual)
		
=== RUN   TestLoadHardwareWithBoardManagerFolderStructure
--- PASS: TestLoadHardwareWithBoardManagerFolderStructure (0.00s)
FAIL
FAIL	arduino.cc/builder/test	0.008s
?   	arduino.cc/builder/types	[no test files]
?   	arduino.cc/builder/utils	[no test files]
PASS
ok  	arduino.cc/properties	0.022s
PASS
ok  	arduino.cc/timeutils	0.003s
```

Could you grep for `FAIL` and check which tests are failing on your build?

facchinm, 2017-01-10:
Ah, I think I got it. It takes 629.807s on your pc to perform a single test, which is more than the 10 minutes timeout. Add `-timeout 60m` or similar to the commandline to fix this. The other FAILS are more difficult to solve without touching the code, unless you prepopulate `src/arduino.cc/builder/test/downloaded_tools/ctags/5.8-arduino11/` folder with your own `ctags` binary, instead than changing its path.

NicoHood, 2017-01-11:
I am already using `go test -timeout 60m arduino.cc/...` so this should not be the problem (and it worked before).

I could symlink the binary to /usr/bin/arduino-ctags but that did not seem to help.

Grep with FAIL:
```
==> Starting check()...
--- FAIL: TestLoadHardware (0.00s)
--- FAIL: TestLoadHardwareMixingUserHardwareFolder (0.00s)
FAIL
FAIL	arduino.cc/builder/test	164.497s
==> ERROR: Build failed, check /var/lib/archbuild/extra-x86_64/arch/build
```

facchinm, 2017-01-11:
These tests are failing because you modified the paths in `platform.txt` 
```
prepare() {
	cd "${srcdir}/${pkgname}-${pkgver}/src/arduino.cc/builder/hardware/"
	sed -i 's#^tools.ctags.path=.*#tools.ctags.path=/usr/bin#' platform.txt
	sed -i 's#^tools.ctags.cmd.path=.*#tools.ctags.cmd.path={path}/arduino-ctags#' platform.txt
}
```

You'd need to find a better way to override that file (or maybe you could simply ignore that errors)

NicoHood, 2017-01-11:
I've moved the prepare() stuff to the package() function. However the ln of the binary gets overwritten by the test tool. So the test actually does not verify that the package works as intended. It would be better to make a changeable platform.txt compatible with the tests.

NicoHood, 2017-02-11:
Any ideas how to fix this? Please reopen the issue as it is still a problem.