arduino/arduino-builder #206: Build now failing when using template class.

Chris--A, 2017-01-20:
Using IDE 1.8.2

After doing up some code for a forum user, I did a test compile. When I came back to my PC, flames were shooting out the CD drive, and the bit bucket under my desk was overflowing with nibbles of 1's.

To solve the issue, I now have to manually add setup and loop prototypes.

```C++
void setup();
void loop();

template< uint16_t nBuffSize >
  class Foo{
    public: 
  
    template< uint16_t N >
      Foo &operator +=( const Foo<N> &ref ){
        //...
        return *this;
    }
};

Foo<64> a;
Foo<32> b;

void setup(){
  a += b;
}

void loop(){}
```

The preprocessor adds the setup and loop prototypes in between:
`template< uint16_t N >` and 
` Foo &operator +=( const Foo<N> &ref ){`
...in the middle of the class.

facchinm, 2017-01-20:
https://github.com/arduino/arduino-builder/pull/182 should solve it, I'm testing it immediately

facchinm, 2017-01-20:
unfortunately it doesn't work in this case :slightly_frowning_face: 
_Investigating_

facchinm, 2017-01-20:
@Chris--A , the "problem" with your code is the return type of  `operator +=` or, more precisely, the fact that there is no space between the ampersand and "operator" word. PR #207 should fix this issue in a sane way (but we should really take a look and refactor the line insertion code)

PR #207 (2017-06-14): Avoid checking if a function uses itself as function pointer
Fixes #206

The sample code from the issue provided a smart way to cheat the dumb FP parser.
In fact, the & is referred to the return type (correct) but the comparator doesn't know about semantics and simply prepends the ampersand before searching for the match.
So the code matches itself, which makes absolutely no sense.
Avoiding this occurrence fixes the issue, however the entire code for prototype line insertion should be refactored in a saner way (see #180 and #191 for another problem)
commit 32825354 (2017-06-14): Avoid checking if a function uses itself as function pointer

Fixes #206

The sample code from the issue provided a smart way to cheat the dumb FP parser.
In fact, the & is referred to the return type (correct) but the comparator doesn't know about semantics and simply prepends the ampersand before searching for the match.
So the code matches itself, which makes absolutely no sense.
Avoiding this occurrence fixes the issue, however the entire code for prototype line insertion should be refactored in a saner way (see #180 and #191 for another problem)
closed 2017-06-14 by facchinm