arduino/arduino-builder #214: Doesn't work if you don't have comments

ShyPixie, 2017-02-10:
I'm trying to understand what's going on here.

Let's get a simple code like this:

```
#include <LedControl.h>

#define clock_pin 11
#define cs_pin 10
#define in_pin 12

LedControl lc = LedControl(in_pin, clock_pin, cs_pin, 1) ;

void setup() {
    lc.shutdown(0,false);
    lc.setIntensity(0,8);
    lc.clearDisplay(0);
}

void loop() {
    lc.setLed(0,0,0,1);
}
```

When compiling and loading this using arduino-builder it doesn't work (No errors, compiles and uploads normally but does not work). But if I write a comment like this:

```
#include <LedControl.h>

#define clock_pin 11
#define cs_pin 10
#define in_pin 12

LedControl lc = LedControl(in_pin, clock_pin, cs_pin, 1) ;

void setup() {
    /*
     * 
     */
    lc.shutdown(0,false);
    lc.setIntensity(0,8);
    lc.clearDisplay(0);
}

void loop() {
    lc.setLed(0,0,0,1);
}

```

Then the code starts to work. I've done several tests and this only happens with the arduino-builder. Using the Arduino IDE works normally. The code generated in these two examples (.hex, .elf) are different, and I think this should not happen.

The command that I've used to build this code is:

```
arduino-builder -hardware /usr/share/arduino-core/hardware
                          -tools  /usr/share/arduino-core/hardware/tools/avr
                          -libraries  /usr/share/arduino-core/libraries
                          -fqbn arduino:avr:uno
                          -verbose -warnings all
                          -build-path build
                          -compile test.ino
```

facchinm, 2017-02-10:
This seems **at least** strange :smile: We'll take a look and report back

facchinm, 2017-02-13:
Ok, maybe I have a hint: when you run through the IDE, the toolchain used is the one bundled by us. Instead, when run from the builder, it is using `avr-gcc` from your PC, which could be incompatible. 
Other problems could be: 
- missing `tools-builder` property
- not adding `-libraries $sketchboot/libraries/`

Please check the difference on the resulting `hex`, not the `elf` file (which changes based on timestamps). I tested it with IDE 1.8.1 (builder 1.3.23) and it appears to work correctly (no problem with or without comments)