arduino/arduino-builder #215: Cannot build for arduino pro mini

uozuAho, 2017-02-21:
I couldn't find the right fqbn to build for my pro mini 3v3 8MHz. I tried arduino:avr:pro, promini, mini to no avail. #78 may have saved me some time here.

It looks like this tool parses 'boards.txt' under ~arduinodir/hardware/arduino/avr. It seems that the problem is the pro mini's settings are nested a few levels deeper to cater for the multiple cpu, voltage and frequency options. For example, pro.menu.cpu.8MHzatmega328. Trying this as a fqbn also didn't work.

A workaround is to add the following text to boards.txt, and using the fqbn arduino:avr:promini328_3v

```
promini328_3v.name=Arduino Pro or Pro Mini

promini328_3v.upload.tool=avrdude
promini328_3v.upload.protocol=arduino
promini328_3v.upload.maximum_size=30720
promini328_3v.upload.maximum_data_size=2048
promini328_3v.upload.speed=57600

promini328_3v.bootloader.tool=avrdude
promini328_3v.bootloader.low_fuses=0xFF
promini328_3v.bootloader.high_fuses=0xDA
promini328_3v.bootloader.extended_fuses=0x05
promini328_3v.bootloader.unlock_bits=0x3F
promini328_3v.bootloader.lock_bits=0x0F
promini328_3v.bootloader.file=atmega/ATmegaBOOT_168_atmega328_pro_8MHz.hex

promini328_3v.build.mcu=atmega328p
promini328_3v.build.f_cpu=8000000L
promini328_3v.build.board=AVR_PRO
promini328_3v.build.core=arduino
promini328_3v.build.variant=eightanaloginputs
```

facchinm, 2017-02-21:
Hi @uozuAho ,
the correct fqbn of boards with multiple menu level is:
`$author:$architecture:$boardname:$property1=$value1:$property2=$value2` , so in your case if will look like
`arduino:avr:pro:cpu=8MHzatmega328`. The documentation is here https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc#options (not very well served by Google, I must admit)

closed 2017-02-21 by facchinm

uozuAho, 2017-02-22:
Thanks for that. Knowing the above and armed with boards.txt, I guess it's not that hard to figure out your hardware's fqbn. I still like the sound of that --list-boards idea from #78 ...