arduino/arduino-builder #222: Library inclusion with extra spaces breaks compilation

smellai, 2017-03-22:
```
//#include < math.h >//math and SPI should be already in your libraries folder by default, don’t worry about these

//#include < SPI.h>

#include < Adafruit_GFX.h >
//we downloaded this just now

#include < Adafruit_PCD8544.h >

Adafruit_PCD8544 display = Adafruit_PCD8544(7, 6, 5, 4, 3); //digital pins used to connect to LCD

#define XPOS 0

#define YPOS 1

float sensorPin = A0; //your sensor pin, you can change this if you want

void setup() 

{

Serial.begin(9600);

display.begin();

}

void loop()

{

float sensorValue = analogRead(sensorPin); //you can change “sensorValue” in all of the code to anything you want

Serial.print("The value of sensor A is: "); //will display this text on the serial monitor

Serial.println(sensorValue); //don’t forget the ln so the next run writes into the next row 

display.clearDisplay(); //clears display each time the loop starts over

display.setCursor(0,0);

display.print("The value of sensor A is: "); //this will be written on the LCD

display.println(sensorValue);

display.display();

delay(1000);

}
```

Compiles fine with:
```
#include <Adafruit_GFX.h>
//we downloaded this just now

#include <Adafruit_PCD8544.h>
```

matthijskooijman, 2017-03-22:
@smellai, could you include the full compilation output (with verbose enabled in the preferences)?

facchinm, 2017-03-22:
In fact, the builder starts including the same folder recursively because it's looking for 
" Adafruit_GFX.h " (with the spaces) but it matches the same string without spaces. I'm setting up a smaller sketch to reproduce the bug

facchinm, 2017-03-22:
Ah, just tested with latest builder and the bug is gone :smile: 
It was probably fixed by https://github.com/arduino/arduino-builder/commit/f7424382e71d27d91aa4e0ec8c66e75fdec1503e or a patch of that series. I believe we can't make a regression test out of it easily (since it should recognize an infinite recursion), but the output is here (for reference only)
```
/ssd/Arduino-master/build/linux/work/arduino-builder -compile -logger=machine -hardware /ssd/Arduino-master/build/linux/work/hardware -hardware /home/martino/.arduino15/packages -hardware /home/martino/eslov-sk/hardware -tools /ssd/Arduino-master/build/linux/work/tools-builder -tools /ssd/Arduino-master/build/linux/work/hardware/tools/avr -tools /home/martino/.arduino15/packages -built-in-libraries /ssd/Arduino-master/build/linux/work/libraries -libraries /home/martino/eslov-sk/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0001 -ide-version=10802 -build-path /tmp/arduino_build_451381 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/ssd/Arduino-master/build/linux/work/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/ssd/Arduino-master/build/linux/work/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/ssd/Arduino-master/build/linux/work/hardware/tools/avr -verbose /tmp/arduino_modified_sketch_92380/sketch_mar22b.ino
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [2.94]
===info ||| Progress {0} ||| [5.88]
===info ||| Using board '{0}' from platform in folder: {1} ||| [uno %2Fssd%2FArduino-master%2Fbuild%2Flinux%2Fwork%2Fhardware%2Farduino%2Favr]
===info ||| Using core '{0}' from platform in folder: {1} ||| [arduino %2Fssd%2FArduino-master%2Fbuild%2Flinux%2Fwork%2Fhardware%2Farduino%2Favr]
===warn ||| Warning: Board {0}:{1}:{2} doesn''t define a ''build.board'' preference. Auto-set to: {3} ||| [MegaCore avr 128 AVR_128]
===warn ||| Warning: Board {0}:{1}:{2} doesn''t define a ''build.board'' preference. Auto-set to: {3} ||| [MegaCore avr 64 AVR_64]
stat /tmp/arduino_modified_sketch_92380/sketch_mar22b.ino: no such file or directory
[martino@arch-elitebook ArduinoCreateAgent-1.0]$ /ssd/Arduino-master/build/linux/work/arduino-builder -compile -logger=machine -hardware /ssd/Arduino-master/build/linux/work/hardware -hardware /home/martino/.arduino15/packages -hardware /home/martino/eslov-sk/hardware -tools /ssd/Arduino-master/build/linux/work/tools-builder -tools /ssd/Arduino-master/build/linux/work/hardware/tools/avr -tools /home/martino/.arduino15/packages -built-in-libraries /ssd/Arduino-master/build/linux/work/libraries -libraries /home/martino/eslov-sk/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0001 -ide-version=10802 -build-path /tmp/arduino_build_451381 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/ssd/Arduino-master/build/linux/work/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/ssd/Arduino-master/build/linux/work/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/ssd/Arduino-master/build/linux/work/hardware/tools/avr -verbose /home/martino/eslov-sk/sketch_mar22b/sketch_mar22b.ino 
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [2.94]
===info ||| Progress {0} ||| [5.88]
===info ||| Using board '{0}' from platform in folder: {1} ||| [uno %2Fssd%2FArduino-master%2Fbuild%2Flinux%2Fwork%2Fhardware%2Farduino%2Favr]
===info ||| Using core '{0}' from platform in folder: {1} ||| [arduino %2Fssd%2FArduino-master%2Fbuild%2Flinux%2Fwork%2Fhardware%2Farduino%2Favr]
===warn ||| Warning: Board {0}:{1}:{2} doesn''t define a ''build.board'' preference. Auto-set to: {3} ||| [MegaCore avr 64 AVR_64]
===warn ||| Warning: Board {0}:{1}:{2} doesn''t define a ''build.board'' preference. Auto-set to: {3} ||| [MegaCore avr 128 AVR_128]
===info ||| Progress {0} ||| [8.82]
===info ||| Progress {0} ||| [11.76]
===info ||| Progress {0} ||| [14.71]
===info ||| Progress {0} ||| [17.65]
===info ||| Progress {0} ||| [20.59]
===info ||| Detecting libraries used... ||| []
===info ||| Progress {0} ||| [23.53]
"/ssd/Arduino-master/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/ssd/Arduino-master/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/ssd/Arduino-master/build/linux/work/hardware/arduino/avr/variants/standard" "/tmp/arduino_build_451381/sketch/sketch_mar22b.ino.cpp" -o "/dev/null"
"/ssd/Arduino-master/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/ssd/Arduino-master/build/linux/work/hardware/arduino/avr/cores/arduino" "-I/ssd/Arduino-master/build/linux/work/hardware/arduino/avr/variants/standard" "-I/home/martino/eslov-sk/libraries/Adafruit_PCD8544_Nokia_5110_LCD_library" "/tmp/arduino_build_451381/sketch/sketch_mar22b.ino.cpp" -o "/dev/null"
"/ssd/Arduino-master/build/linux/work/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/ssd/Arduino [... cut ...] 110_LCD_library" "-I/home/martino/eslov-sk/libraries/Adafruit_PCD8544_Nokia_5110_LCD_library" "-I/home/martino/eslov-sk/libraries/Adafruit_PCD8544_Nokia_5110_LCD_library" "-I/home/martino/eslov-sk/libraries/Adafruit_PCD8544_Nokia_5110_LCD_library" "-I/home/martino/eslov-sk/libraries/Adafruit_PCD8544_Nokia_5110_LCD_library" "-I/home/martino/eslov-sk/libraries/Adafruit_PCD8544_Nokia_5110_LCD_library" "/tmp/arduino_build_451381/sketch/sketch_mar22b.ino.cpp" -o "/dev/null"
.... continues forever
```

closed 2017-03-22 by facchinm