arduino/arduino-builder #227: How to upload compiled sketch?

stormbytes, 2017-05-18:
This isn't really an issue but fair question I believe.

How do you upload the compiled sketch?

facchinm, 2017-05-18:
Right now the upload procedure is still performed by the java IDE (or the Create agent), so you need to retrieve the suitable `upload` commandline (for example https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt#L105), manually replace the placeholders and execute it on a shell. 
In the near future we are evaluating to port the upload procedure to a standalone tool so I'll post here as soon as things start moving.

closed 2017-05-18 by facchinm

stormbytes, 2017-05-18:
After some hours I arrived at the very same conclusion!

Posting this resource on how to run avrdude from the command line (or rather, how to figure out how to run avrdude etc)

http://stackoverflow.com/questions/23780125/uploading-arduino-hex-file-using-avrdude