arduino/arduino-builder #231: Compiler changes causes deletion of sketch

seware74, 2017-06-07:
Windows 10 - 64 bit
Version 1.8.2

If I change the compiler options in the platform.txt file, the next build of the sketch deletes the entire contents of the source folder. 

This occurs on execution of the second command in the build process: 

""C:\Program Files (x86)\Arduino\arduino-builder" -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Steve\Documents\Arduino\libraries" -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10802 -build-path "C:\Users\Steve\Documents" -warnings=all -build-cache "C:\Users\Steve\AppData\Local\Temp\arduino_cache_521285" -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path="C:\Program Files (x86)\Arduino\hardware\tools\avr" -prefs=runtime.tools.arduinoOTA.path="C:\Program Files (x86)\Arduino\hardware\tools\avr" -prefs=runtime.tools.avrdude.path="C:\Program Files (x86)\Arduino\hardware\tools\avr" -verbose -debug-level=10 "C:\Users\Steve\Documents\GitHub\Firmware\cnc_ctrl_v1\cnc_ctrl_v1.ino""

Debug output from the offending line:

===debug ||| Ts: {0} - Running: {1} ||| [1496850676 GenerateBuildPathIfMissing]
===info ||| Progress {0} ||| [0.00]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 EnsureBuildPathExists]
===info ||| Progress {0} ||| [2.94]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 ContainerSetupHardwareToolsLibsSketchAndProps]
===info ||| Progress {0} ||| [5.88]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 AddAdditionalEntriesToContext]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 FailIfBuildPathEqualsSketchPath]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 HardwareLoader]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 PlatformKeysRewriteLoader]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 RewriteHardwareKeys]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 ToolsLoader]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 TargetBoardResolver]
===info ||| Using board '{0}' from platform in folder: {1} ||| [mega C%3A%5CProgram+Files+%28x86%29%5CArduino%5Chardware%5Carduino%5Cavr]
===info ||| Using core '{0}' from platform in folder: {1} ||| [arduino C%3A%5CProgram+Files+%28x86%29%5CArduino%5Chardware%5Carduino%5Cavr]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 AddBuildBoardPropertyIfMissing]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 LibrariesLoader]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 SketchLoader]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 SetupBuildProperties]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 LoadVIDPIDSpecificProperties]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 SetCustomBuildProperties]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 AddMissingBuildPropertiesFromParentPlatformTxtFiles]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 ContainerBuildOptions]
===info ||| Progress {0} ||| [8.82]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 CreateBuildOptionsMap]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 LoadPreviousBuildOptionsMap]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 WipeoutBuildPathIfBuildOptionsChanged]
===info ||| Build options changed, rebuilding all ||| []
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 StoreBuildOptionsMap]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 WarnAboutPlatformRewrites]
===info ||| Progress {0} ||| [11.76]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 RecipeByPrefixSuffixRunner]
===info ||| Progress {0} ||| [14.71]
===debug ||| Looking for recipes like {0}*{1} ||| [recipe.hooks.prebuild .pattern]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 ContainerMergeCopySketchFiles]
===info ||| Progress {0} ||| [17.65]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 SketchSourceMerger]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 SketchSaver]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 AdditionalSketchFilesCopier]
===debug ||| Ts: {0} - Running: {1} ||| [1496850676 PrintUsedAndNotUsedLibraries]
===debug ||| Ts: {0} - Running: {1} ||| [1496850677 PrintUsedLibrariesIfVerbose]
===debug ||| Ts: {0} - Running: {1} ||| [1496850677 Sizer]
open C:\Users\Steve\Documents\GitHub\Firmware\cnc_ctrl_v1\Axis.cpp: The system cannot find the path specified.
*os.PathError open C:\Users\Steve\Documents\GitHub\Firmware\cnc_ctrl_v1\Axis.cpp: The system cannot find the path specified.
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/i18n/errors.go:19 (0x4afd04)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/additional_sketch_files_copier.go:68 (0x493a5a)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_merge_copy_sketch_files.go:48 (0x498b89)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:191 (0x494a42)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:124 (0x4944af)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:222 (0x495035)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/arduino-builder/main.go:338 (0x402f80)
/home/jenkins/go/src/runtime/proc.go:183 (0x42b90e)
/home/jenkins/go/src/runtime/asm_386.s:1612 (0x4520c1)

facchinm, 2017-06-09:
Hi @seware74 ,
the wipe of the build directory is totally expected, since the old behaviour was buggy (in fact, if you changed anything in platform.txt the changes were not applied to the resulting binary since the old cached file were considered valid). 
What you are experiencing is quite different and related to the line ```-build-path "C:\Users\Steve\Documents" ```,. In fact, the build directory should be a "temporary" folder (even if it's on the hard disk and not in RAM) because any change you made to the configuration (also changing target board) will result in a wipe of the entire directory, even if it contains other files.

seware74, 2017-06-09:
I guess i understand... but it's not everything in the build path that was deleted.  My code was at c:\users\steve\documents\github\firmware
Everything in the the github folder was deleted, which is the level between the build path and source folder. 
You Might consider adding a warning somewhere.  Imagine if the build path was c:    yikes!Or better yet don't bury the output in a temp location,  forcing folks to modify the build path to find the output easily.
Upgraded to 1.8.3 yesterday and it does not have this behaviour. 
Doesnt matter to me,  I switched to atmel studio so I could debug on target anyway. 
Thanks for the reply

  sketch (#231)

matthijskooijman, 2017-06-11:
I wonder if you manually configured a `build.path` in `preferences.txt` or using the `--pref` option to Arduino? A previous similar report turned out to be caused by an old (and forgotten) change to this preference.

per1234, 2021-10-14:
As explained, clearing the build folder is the expected behavior. We have a dedicated issue about documenting the behavior here:
https://github.com/arduino/Arduino/issues/11672

closed 2021-10-14 by per1234