arduino/arduino-builder #235: Setting build-path to nested subdirectory

GeorgeHahn, 2017-06-15:
Related to #4 - if the build dir is set to a folder within the working directory (eg `./build`), a recursive folder structure is created (see #4 for an example).

Issue #4 notes that an error message was added to guard against this behavior. This error message should be extended to cover any folder contained within the working directory.

matthijskooijman, 2017-06-15:
Hm, I thought we already fixed this and allowed a build directory within the sketch directory. If not, then it's certainly on my list of things that should be made possible :-)

luni64, 2018-11-21:
Any news on that? It still get an error when I define the build directory in the sketch directory
`Rel: can't make C:\Users\lutz\source\newProject(1)\bin\core\core.a relative to bin`
Any workaround?

facchinm, 2018-11-21:
@luni64 would you mind attaching the full compiler output? The problem looks a bit Windows specific

facchinm, 2018-11-21:
Thanks for the log.
I think there might be a workaround by using an absolute path for `-build-path=` (I would write `$PWD/bin` on Linux). The piece of code that kicks in is https://github.com/arduino/arduino-builder/blob/bbf705133803b0825818af1ed51cf694eef17fc7/phases/linker.go#L58-L61
For some reason `ctx.BuildPath` is still relative in that snippet; I'll take a look tomorrow to understand what is happening :slightly_smiling_face:

facchinm, 2018-11-21:
@luni64 I just pushed a test PR here https://github.com/arduino/arduino-builder/pull/308 , as soon as the binary is ready it would be cool if you could test it in your environment!

luni64, 2018-11-21:
Sure, I'll test it as soon as it is available

luni64, 2018-11-21:
That didn't work, refuses to compile at all now. Here the log:
```
> Executing task: & 'C:/Program Files/arduino-1.8.5/arduino-builder' -verbose=1 -logger=human '-hardware= C:\Program Files\arduino-1.8.5\hardware' -build-path=bin '-tools=C:\Program Files\arduino-1.8.5\tools-builder' '-tools=C:\Program Files\arduino-1.8.5\hardware\tools\avr' -fqbn=teensy:avr:teensy31:usb=serial,speed=96,opt=o2std,keys=en-us src\main.cpp <

FindFirstFile C:\Users\lutz\source\newProject(1)\ C:\Program Files\arduino-1.8.5\hardware: Die Syntax für den Dateinamen, Verzeichnisnamen oder die Datenträgerbezeichnung ist falsch.
The terminal process terminated with exit code: 1
```

facchinm, 2018-11-22:
@luni64 there's a space between `-hardware=` and the actual path; I think one of the patches we merged for the commandline (removing the  ', ' as separator) made the builder become a bit more picky. May you retest without the space? Thanks!

luni64, 2018-11-22:
Thanks, that fixed it. Works perfectly now.

facchinm, 2018-11-23:
Super, thanks.
The extra hex is `with_bootloader` ? That thing is terribly broken right now, should be fixed by https://github.com/arduino/arduino-builder/pull/295 . 

Instead, if you refer to the real boardID (like `uno` or similar), that step is handled by the java IDE using "Sketch/Export compiled binary" menu, and the boardID follows board's `variant` (you can find this information in `boards.txt` for the given core).

booniiiel, 2018-11-23:
hi, I got the similar problem My arduino IDE need to be closed and open again in order to upload the code to arduino uno. Before this, i had edited the path in preference, does it related?
The error I got:

Rel: can't make C:\Users\UserName\AppData\Local\Temp\arduino_cache_239826\core\core_arduino_avr_uno_0c812875ac70eb4a9b385d8fb077f54c.a relative to D:\Arduino\hex
Error compiling for board Arduino/Genuino Uno.

facchinm, 2018-11-23:
Yes, they are related; the build folder should be in the same partition of the IDE installation. It is a windows specific problem :disappointed: We must shorten the commandline and paths because of other win limitations, and we achieve this by making paths relative. 
But on win, two paths on different partitions can't be relative. 

I'd suggest to avoid moving the default build folder and use "Sketch/Export Compiled Binary" menu in case you need to save the final hex.

booniiiel, 2018-11-24:
> Yes, they are related; the build folder should be in the same partition of the IDE installation. It is a windows specific problem 😞 We must shorten the commandline and paths because of other win limitations, and we achieve this by making paths relative.
> But on win, two paths on different partitions can't be relative.
> 
> I'd suggest to avoid moving the default build folder and use "Sketch/Export Compiled Binary" menu in case you need to save the final hex.

i see.. so in order to get the arduino back to normal, i need to delete the build path that i created?

facchinm, 2018-11-26:
Just remove `build.path` entry in `preferences.txt` . This will tell the IDE to automatically create a correct temporary build directory in the "right" path.

facchinm, 2019-01-03:
Closing as fixed

closed 2019-01-03 by facchinm