arduino/arduino-builder #238: "Archiving built core" message should only be shown in verbose output and in white, not red

per1234, 2017-06-23:
Using Arduino IDE 1.8.3 on Windows 7 64 bit

The message:
```
Archiving built core (caching) in: C:\Users\per\AppData\Local\Temp\arduino_cache_659403\core\core_arduino_avr_uno_d0519d097723f7476aa49015be44facf.a
```
is shown in red in the non-verbose compilation output. It should only be shown in the verbose compilation output and the color should be white as it is not a warning or error.

As demonstrated by:
http://forum.arduino.cc/index.php?topic=485289
This message can cause undue concern for users or mislead them from the true cause of their problem, yet it provides no useful information to the vast majority of users.

facchinm, 2017-06-23:
@per1234  I agree with it, thanks for reporting. I probably never noticed it since I always compile with max verbosity :smile:  Fixing right now!

facchinm, 2017-06-23:
Fixed by https://github.com/arduino/arduino-builder/commit/7883e01702e70a617ee708dddf9de9884d89b578

matthijskooijman, 2017-12-07:
The fix commit made the message only appear in verbose mode, but failed to make it appear in white, so I'm reopening this issue. I just submitted a PR to fix this.

closed 2017-12-11 by facchinm

sanjeewasam, 2018-04-01:
What is the status on this

sanjeewasam, 2018-04-01:
So having red without getting an error OK?

facchinm, 2018-04-03:
@sanjeewasam the line is printed in white in [Beta channel](https://www.arduino.cc/en/Main/Software#beta) at the moment, if you could test it and report if you find any bug it would be awesome!

narjes1374, 2018-04-21:
hello in programing ESP i have error i don t know this error 

Archiving built core (caching) in: C:\Users\narjes\AppData\Local\Temp\arduino_cache_348130\core\core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____8f9b38d3d6da3b10a73a5ebc46da2c3b.a
Sketch uses 231953 bytes (53%) of program storage space. Maximum is 434160 bytes.
Global variables use 32312 bytes (39%) of dynamic memory, leaving 49608 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

narjes1374, 2018-04-21:
what meaning of verbose mode?

lprasanth, 2018-05-21:
Hello Iam  trying  hard to solve the  error did you find any solutions ?

commit 7883e017 (2017-06-23): Let "archiving core" message appear only if verbose

Fixes #238
commit 6ec99626 (2017-12-11): Print "archiving core" message at INFO level

This previously used the DEBUG level, which is shown in red by the
Arduino IDE. Commit 7883e01 (Let "archiving core" message appear only if
verbose) attempted to fix this, but did not.

While here, this also changes Fprintln with an explicit Stdout argument
to a regular Println, since Stdout is the default anyway.

This commit properly fixes #238.