arduino/arduino-builder #240: valid fqdn from example fails

gherlein, 2017-08-01:
Using a copy of the blink.ino example I cannot get arduin-builder to work:

arduino-builder -build-path ./build  -hardware ~/Library/Arduino15/packages/arduino/hardware/ -hardware ~/Library/Arduino15/packages/SparkFun/hardware/ -tools ~/Library/Arduino15/packages/arduino/tools/ -tools ~/Library/Arduino15/packages/SparkFun/tools/ -libraries ~/herlein/src/arduino/libraries/ -libraries ./libraries -fqbn arduino:avr:uno -verbose -debug-level 10 blink.ino
Ts: 1501598754 - Running: GenerateBuildPathIfMissing
Ts: 1501598754 - Running: EnsureBuildPathExists
Ts: 1501598754 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1501598754 - Running: AddAdditionalEntriesToContext
Ts: 1501598754 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1501598754 - Running: HardwareLoader
Ts: 1501598754 - Running: PlatformKeysRewriteLoader
Ts: 1501598754 - Running: RewriteHardwareKeys
Ts: 1501598754 - Running: ToolsLoader
Ts: 1501598754 - Running: TargetBoardResolver
Ts: 1501598754 - Running: PrintUsedAndNotUsedLibraries
Ts: 1501598754 - Running: PrintUsedLibrariesIfVerbose
Ts: 1501598754 - Running: Sizer
arduino: Unknown package
*errors.errorString arduino: Unknown package
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/i18n/errors.go:12 (0xb3adb)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/target_board_resolver.go:59 (0xad7a4)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_setup.go:59 (0x9bab3)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:191 (0x976ad)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:124 (0x9702c)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:222 (0x97c95)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/arduino-builder/main.go:338 (0x379c)
/home/jenkins/go/src/runtime/proc.go:183 (0x2c0a4)
/home/jenkins/go/src/runtime/asm_amd64.s:2086 (0x582d1)

make: *** [all] Error 1

I am assuming that the error is somewhere in the "-fqbn arduino:avr:uno" 

What am I doing wrong?

NOTE:  yes, this works from the IDE.  Yes, I searched the web for an answer first.

facchinm, 2017-08-01:
Hi @gherlein ,
are you using the same `arduino-builder` binary shipped with the IDE? Could you also paste the verbose output from the IDE itself when you compile the same sketch? Thanks

gherlein, 2017-08-01:
Yes, same one as shipped with IDE.  Looks like it is using different hardware folders at least...

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/gherlein/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/gherlein/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/gherlein/herlein/src/arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10803 -build-path /var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826 -warnings=none -build-cache /var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_cache_849820 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/gherlein/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=/Users/gherlein/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.1.1 -prefs=runtime.tools.avrdude.path=/Users/gherlein/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/gherlein/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/gherlein/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/gherlein/herlein/src/arduino/libraries -fqbn=arduino:avr:uno -vid-pid=0X2341_0X0043 -ide-version=10803 -build-path /var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826 -warnings=none -build-cache /var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_cache_849820 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/gherlein/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.arduinoOTA.path=/Users/gherlein/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.1.1 -prefs=runtime.tools.avrdude.path=/Users/gherlein/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board 'uno' from platform in folder: /Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18
Using core 'arduino' from platform in folder: /Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18
Detecting libraries used...
"/Users/gherlein/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18/cores/arduino" "-I/Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18/variants/standard" "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/gherlein/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18/cores/arduino" "-I/Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18/variants/standard" "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/sketch/Blink.ino.cpp" -o "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Users/gherlein/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18/cores/arduino" "-I/Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/1.6.18/variants/standard" "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/sketch/Blink.ino.cpp" -o "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"/Users/gherlein/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Users/gherlein/Library/Arduino15/packages/arduino/hardware/avr/ [... cut ...] o.eep"
"/Users/gherlein/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -R .eeprom  "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/Blink.ino.elf" "/var/folders/vt/qg0v_q5n5k99pnp_yg24c7s80000gn/T/arduino_build_678826/Blink.ino.hex"
Sketch uses 928 bytes (2%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

gherlein, 2017-08-01:
Trimming things down and matching what the IDE does:

arduino-builder -build-path ./build  -hardware ~/Library/Arduino15/packages -tools ~/Library/Arduino15/packages -libraries ~/herlein/src/arduino/libraries -fqbn arduino:avr:uno -verbose -debug-level 10 blink.ino

Seems to find the packages but now spits out a missing ctags error:

ctags pattern is missing
*errors.errorString ctags pattern is missing
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/i18n/errors.go:12 (0xb3adb)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/ctags_runner.go:55 (0x9bf41)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_add_prototypes.go:56 (0x9819e)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:191 (0x976ad)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:124 (0x9702c)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:222 (0x97c95)
/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/arduino-builder/main.go:338 (0x379c)
/home/jenkins/go/src/runtime/proc.go:183 (0x2c0a4)
/home/jenkins/go/src/runtime/asm_amd64.s:2086 (0x582d1)

make: *** [all] Error 1

gherlein, 2017-08-01:
I solved it by carefully matching the folders that the IDE used.  I will try to add to the docs for the next person and submit a PR.

closed 2017-08-01 by gherlein