arduino/arduino-builder #241: need more documentation on -fqbn param

gherlein, 2017-08-01:
How does one construct the fqbn param?  

How does one get a list of boards possible?

NOTE:  this is similar to the --list-boards enhancement request earlier, but short of getting that can we at least get some examples and/or docs?

Amertz08, 2018-01-18:
Ran into this issue too. Found the following.

https://forum.arduino.cc/index.php?topic=376923.0

I'm on Mac so for me it looks like from the root of my hardware folder `/Applications/Arduino.app/Contents/Java/hardware/` I build my `fqbn` by looking at the folder and then the target in `boards.txt` so for me `PATH_TO_HARDWARE/arduino/avr/boards.txt` with a desired target of `Arduino/Genuino Uno` became `arduino:avr:uno`

Hope that helps and makes sense. But yeah documentation on that is lacking.

Adam

adumont, 2018-12-21:
Indeed.
My trick? Use Visual Studio Code, with the Arduino Extension. When you configure the board for sketch, it will create a new file in your sketch directory, under .vscode/arduino.json containing at least these three fields: sketch, board and configuration.

Here is an example:

```
{
    "sketch": "ESP-sc-gway.ino",
    "board": "esp8266:esp8266:d1_mini",
    "configuration": "CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=sdk,UploadSpeed=921600"
}
```

When building with VSCode, I noticed that it uses arduino-builder, and the -fqbn is `esp8266:esp8266:d1_mini:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=sdk,UploadSpeed=921600`, which is the contatenation of `board` and `configuration` values.

At least that works for me :)

jjtmp, 2022-08-20:
> Ran into this issue too. Found the following.
> 
> https://forum.arduino.cc/index.php?topic=376923.0

Close. When you go to the link from @Amertz08 you get the hint to the redirect to https://arduino.github.io/arduino-cli/0.26/platform-specification/

The latter link is better than the former, I suppose. When you search for "fqbn" on the page you find even variants of providing additional details.

per1234, 2022-09-27:
Thanks @jjtmp for sharing the link to the platform specification. There is also some information about FQBN in the FAQ:

https://arduino.github.io/arduino-cli/latest/FAQ/#whats-the-fqbn-string

**arduino-builder** is deprecated in favor of [**Arduino CLI**](https://arduino.github.io/arduino-cli/latest/) so no further documentation will be developed specific to **arduino-builder**, but it would be beneficial to improve the documentation about FQBNs in general. We are using https://github.com/arduino/arduino-cli/issues/355 to track that task. I will close this issue in favor of that one. If anyone has additional information to share on the subject, feel free to comment in the other thread.

closed 2022-09-27 by per1234