arduino/arduino-builder #244: Can't install

webdev23, 2017-08-27:
Seriously whoever say that everyone know how to manage go-lang ... should go back to his java paradigm, definetly.

Please clarify your docs, mostly for the installation procedures, or better, make a one-click package installer.
There is hundreds of ways to make it simple, so why don't do it?

> 
>  go build arduino.cc/arduino-builder
> can't load package: package arduino.cc/arduino-builder: cannot find package "arduino.cc/arduino-builder" in any of:
> 	/snap/go/727/src/arduino.cc/arduino-builder (from $GOROOT)
> 	/home/zen/go/src/arduino.cc/arduino-builder (from $GOPATH)

matthijskooijman, 2017-08-28:
Not sure what the problem is exactly. If I follow the instructions (except that I do not run setup_go_vars entirely, just set GOPATH, we should probably improve that part), it builds correctly:

```
matthijs@grubby:/tmp$ git clone https://github.com/arduino/arduino-builder.git
Cloning into 'arduino-builder'...
remote: Counting objects: 4953, done.
remote: Total 4953 (delta 0), reused 0 (delta 0), pack-reused 4953
Receiving objects: 100% (4953/4953), 1.14 MiB | 977.00 KiB/s, done.
Resolving deltas: 100% (3131/3131), done.
matthijs@grubby:/tmp$ cd ar=
-bash: cd: ar=: No such file or directory
matthijs@grubby:/tmp$ cd arduino-builder/
matthijs@grubby:/tmp/arduino-builder$ export GOPATH=$(pwd)
matthijs@grubby:/tmp/arduino-builder$ go get github.com/go-errors/errors
matthijs@grubby:/tmp/arduino-builder$ go get github.com/stretchr/testify
matthijs@grubby:/tmp/arduino-builder$ go get github.com/jstemmer/go-junit-report
matthijs@grubby:/tmp/arduino-builder$ go build arduino.cc/arduino-builder
matthijs@grubby:/tmp/arduino-builder$ ls
arduino-builder  bin  CONTRIBUTING.md  fmt_fix_vet  LICENSE.txt  pkg  README.md  setup_go_env_vars  src
```

You'll have to provide more details about how you set things up. Also, I would suggest you reconsider your (even slightly) snarky tone, which seems inappropriate to me.

webdev23, 2017-08-28:
You are right. Please forgive. 
The procedure was failing without the  **export GOPATH=$(pwd)** line that you just shown.
(ubuntu64 4.4.0-92-generic )

> go version
> go version go1.9 linux/amd64u

> zen@book:~$ git clone https://github.com/arduino/arduino-builder.git
> Cloning into 'arduino-builder'...
> remote: Counting objects: 4953, done.
> remote: Total 4953 (delta 0), reused 0 (delta 0), pack-reused 4953
> Receiving objects: 100% (4953/4953), 1.14 MiB | 797.00 KiB/s, done.
> Resolving deltas: 100% (3131/3131), done.
> Checking connectivity... done.
> zen@book:~$ cd arduino-builder
> zen@book:~/arduino-builder$ ls
> CONTRIBUTING.md  fmt_fix_vet  LICENSE.txt  README.md  setup_go_env_vars  src
> zen@book:~/arduino-builder$ export GOPATH=$(pwd)
> zen@book:~/arduino-builder$ go get github.com/go-errors/errors
> zen@book:~/arduino-builder$ go get github.com/stretchr/testify
> zen@book:~/arduino-builder$ go get github.com/jstemmer/go-junit-report
> zen@book:~/arduino-builder$ go build arduino.cc/arduino-builder
> zen@book:~/arduino-builder$ ls
> arduino-builder  CONTRIBUTING.md  LICENSE.txt  README.md          src
> bin              fmt_fix_vet      pkg          setup_go_env_vars
> 
> 

So, thanks.
The tool is also included in the arduino 1.8.4 package.
This tool is finally just a wrapper script for avrdude? Isn'it?

matthijskooijman, 2017-08-29:
No, `avrdude` (which handles uploading) is directly called by the Arduino IDE. `arduino-builder` is in charge of calling `avr-gcc` and some other tools to compile a sketch. Since this is fixed for you, I'll mark this issue as closed. I've just created #245 with some improvements to the README which might have prevented your issue, feel free to see what you think of them.

closed 2017-08-29 by matthijskooijman