arduino/arduino-builder #253: dot_a_linkage ignored when no src folder

PaulStoffregen, 2017-10-24:
The dot_a_linkage=true option in library.properties appears to be ignored for libraries without a src folder, where their code resides in the main folder.

facchinm, 2017-10-30:
I took another look at this, and the behaviour seems just fine since the `dot_a_linkage` option only applies to library specification > = 1.5, which specifies that the code must reside in `src` folder. I'd close it as WONTFIX, feel free to comment if there is any particular reason why this behaviour should be changed.

closed 2017-10-30 by facchinm

PaulStoffregen, 2017-10-30:
Ok then.  Not sure I quite see how the spec says this (the rest of the library.properties stuff applies to both directory structures), but hardly seems worth arguing.  If this is the intended behavior, might be worthwhile to mention in the dot_a_linkage documentation.

per1234, 2017-10-30:
>might be worthwhile to mention in the dot_a_linkage documentation.

I have done so:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification/_compare/5e5c3c2fe76662e13b8099c7a72004a5c6e5344d...7e773c7f19154702bf15da53dc3baa26815b3091

PaulStoffregen, 2017-10-30:
ok