arduino/arduino-builder #254: Compilation fails with OneDrive / "Cloud reparse points"

matthijskooijman, 2017-11-16:
It seems recent Windows 10 versions combined with OneDrive suffer from arduino-builder not being able to read files. This is reported as arduino/Arduino#6867 and arduino/Arduino#6932. The symptoms are similar as #90, but I think the underlying issue is similar but different, so I'm opening a new issue.

arduino/Arduino#6932 contains the error output:

```
    Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

    C:\Apps\arduino-latest\arduino-builder -dump-prefs -logger=machine -hardware C:\Apps\arduino-latest\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -hardware C:\Users\HaMac\OneDrive\Documents\Ardunio\hardware -tools C:\Apps\arduino-latest\tools-builder -tools C:\Apps\arduino-latest\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries C:\Apps\arduino-latest\libraries -libraries C:\Users\HaMac\OneDrive\Documents\Ardunio\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_721501 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino

    readlink C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino: The system cannot find the file specified.

    Error compiling for board Arduino/Genuino Uno.
```

The reporter has tested a number of versions, and has found that 1.6.5 works, but 1.6.6 breaks things, on this particular system. The latter version is the one introducing arduino-builder.

One difference here, is that the .ino file seems to be a link (or some other special file), not the directory.

@phil85521 @ngmacha, could either of you get some more info on the .ino file that is failing? E.g. look at its properties or use some other Windows tool to figure out what kind of file it is, whether it is indeed a junction or some other kind of special file? I don't have any specific ideas what to do, not being a Windows-user.

Looking at the go code previously patched (https://github.com/golang/go/commit/571738569e9b6bb30ebceaf2eb5936f6ee254e34#diff-495e6a9c70ec790e4be07ee30c55141f) it seems that this file might be (or have?) some kind of "reparse point" that is not a symlink or junction (which are the supported types). This [post](https://stackoverflow.com/questions/47264992/one-drive-as-reparse-point) suggests that OneDrive uses the "Cloud" reparse tag (`IO_REPARSE_TAG_CLOUD_7`), which golang does not seem to explicitly support (in its [master branch](https://github.com/golang/go/blob/master/src/syscall/syscall_windows.go#L1022-L1055)).

I suspect this is something to fix at the golang end, it seems there is already a related issue open at https://github.com/golang/go/issues/22579 (which I'll provide with some extra info next).

matthijskooijman, 2017-11-16:
On the [forum](http://forum.arduino.cc/index.php?topic=509222.msg3485724#msg3485724), a workaround is suggested:

> I had the same problem with onedrive after the fall creator update. Just un-check "Files On-Demand" in onedrive setting, arduino IDE can normally read and compile sketch.  In my case, it works.

rcarmo, 2017-11-16:
That workaround doesn’t apply in all cases. I’ve yet to figure out why, though.

ssylvan, 2017-11-16:
@rcarmo I had to disable the "files on demand" feature, and then copy my files to a new folder in onedrive before it would work. It seems that if a given file was ever "on demand" it stays in symlink mode (forever? for a while? no idea).

rcarmo, 2017-11-16:
That’s because it creates a file system junction. However, Windows 10 creates dozens of junctions for housekeeping purposes, so moving your files to another folder doesn’t necessarily mean you’ll land in a junction-free zone... and I, for one, really need to have his fixed “right”...

matthijskooijman, 2017-11-17:
@ngmacha, could you try running this command (it's the failing command, but with more verbosity):

```
C:\Apps\arduino-latest\arduino-builder -dump-prefs -logger=machine -hardware C:\Apps\arduino-latest\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -hardware C:\Users\HaMac\OneDrive\Documents\Ardunio\hardware -tools C:\Apps\arduino-latest\tools-builder -tools C:\Apps\arduino-latest\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries C:\Apps\arduino-latest\libraries -libraries C:\Users\HaMac\OneDrive\Documents\Ardunio\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_721501 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino -debug-level 20
```

I believe the next step is to file a proper bugreport with go (the one I linked before is a related, but different issue), along with a minimal program to reproduce the problem. I've looked around the source code to figure out what arduino-builder is doing exactly, but I couldn't quite figure out what code path is triggering this problem exactly. We're not calling `Readlink` directly and the only thing in the go source that I could find that calls it is the `WalkSymlinks` stuff, which I think we're not using. Hopefully the output of the above command will provide some hints about where too look (since it seems the error returned is not an actual error object with a backtrace we could print, but is just a plain error message ([generated here](https://github.com/golang/go/blob/187957d37056592203fd758ae0245a28f4518122/src/os/file_posix.go#L23)).

I'm short on time to dig into this further, so if anyone else can manage to create a small representative testcase and report a bug with go, please go ahead!

ngmacha, 2017-11-17:
@matthijskooijman Here is the output (I had to edit the command to point to the correct Temp\arduino_build path, as since my previous post I've reinstalled Arduino 1.8.5 on my system, so some paths are changed):
> C:\Users\HaMac>C:\Apps\arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Apps\arduino\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -hardware C:\Users\HaMac\OneDrive\Documents\Ardunio\hardware -tools C:\Apps\arduino\tools-builder -tools C:\Apps\arduino\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries C:\Apps\arduino\libraries -libraries C:\Users\HaMac\OneDrive\Documents\Ardunio\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_531879 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino -debug-level 20
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [33.33]
GetFileAttributesEx C:\Users\HaMac\OneDrive\Documents\Ardunio\hardware: The system cannot find the path specified.

When I remove that offending -hardware option, it outputs this:
> C:\Users\HaMac>C:\Apps\arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Apps\arduino\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -tools C:\Apps\arduino\tools-builder -tools C:\Apps\arduino\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries C:\Apps\arduino\libraries -libraries C:\Users\HaMac\OneDrive\Documents\Ardunio\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_531879 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino -debug-level 20
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [33.33]
===info ||| Using board '{0}' from platform in folder: {1} ||| [uno C%3A%5CUsers%5CHaMac%5CAppData%5CLocal%5CArduino15%5Cpackages%5Carduino%5Chardware%5Cavr%5C1.6.20]
===info ||| Using core '{0}' from platform in folder: {1} ||| [arduino C%3A%5CUsers%5CHaMac%5CAppData%5CLocal%5CArduino15%5Cpackages%5Carduino%5Chardware%5Cavr%5C1.6.20]
open C:\Users\HaMac\OneDrive\Documents\Ardunio\libraries: The system cannot find the path specified.

Then, finally, removing the -library option above:
>C:\Users\HaMac>C:\Apps\arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Apps\arduino\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -tools C:\Apps\arduino\tools-builder -tools C:\Apps\arduino\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries C:\Apps\arduino\libraries -fqbn=arduino:avr:uno -ide-version=10805 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_531879 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino -debug-level 20
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [33.33]
===info ||| Using board '{0}' from platform in folder: {1} ||| [uno C%3A%5CUsers%5CHaMac%5CAppData%5CLocal%5CArduino15%5Cpackages%5Carduino%5Chardware%5Cavr%5C1.6.20]
===info ||| Using core '{0}' from platform in folder: {1} ||| [arduino C%3A%5CUsers%5CHaMac%5CAppData%5CLocal%5CArduino15%5Cpackages%5Carduino%5Chardware%5Cavr%5C1.6.20]
GetFileAttributesEx C:\Users\HaMac\OneDrive\Documents\Ardunio\BlinkTwoLEDs\BlinkTwoLEDs.ino: The system cannot find the path specified.

As only the sketch itself remain as offender, nothing left to remove...
So actually in all cases the GetFileAttributesFx was unable to find the folder/file on the OneDrive. Don't know if this is of any help. And as I don't have any experience with go, guess I can't help with creating the case too...

matthijskooijman, 2017-11-17:
Thanks!. It's interesting that the error message changed somewhat: It's `Open` and `GetFileAttributesEx` that are complaining now, no longer `ReadLink`.

jlvandusen, 2017-12-17:
I concur - on Windows 10 fall and one drive this started occurring I have to copy the sketches out of onedrive and compile to test now... this is not acceptable as most builders use the onedrive for their My Documents to store their data in the cloud/backup etc.

facchinm, 2017-12-18:
Fix is scheduled for Go 1.11 (or 1.10 since also Docker is affected); we'll recompile the builder as soon as an go updated toolchain is out.

matthijskooijman, 2017-12-18:
@facchinm do you have a link for that info? The go bug report I previously linked doesn't say this, so is there another report you're tracking?

facchinm, 2018-01-25:
https://go-review.googlesource.com/c/go/+/86556 targets the issue, but we must wait mid-2018 for it to land on a released version of golang (https://github.com/golang/go/issues/22579#issuecomment-355807833) . If the number of affected users becomes broader we could backport the patch and recompile the Windows target but otherwise I'd wait for the release.

species5618, 2018-02-25:
i have a work around
place the arduino folder outside of the onedrive structure and use a symbolic link inside the ondrive to folder to reference it,

arduino IDE work and it syncs on one drive

i use it to have my gdrive flolder copied to onedrive too

https://www.howtogeek.com/howto/16226/complete-guide-to-symbolic-links-symlinks-on-windows-or-linux/

mklink /D C:\Users\species5618\OneDrive\Documents\Arduino c:\stuff\Arduino

ngmacha, 2018-02-27:
Yes, @species5618 this is actually a great workaround!! And it works for Energia too! Thank you very much.

dmikedaniel, 2018-04-11:
Arduino was working fine on my PC until the latest update ( or so it seems).  Now I get 

readlink C:\Users\dmike\OneDrive\Documents\Arduino\Turn_It_Around\Turn_It_Around.ino: The system cannot find the file specified.

Error compiling for board Arduino/Genuino Uno.

I've tried doing a clean uninstall and reinstall.  I can verify and compile a new empty sketch or an example.  I might be able to 

I'm a little lost when reading some of these comments, but some of them I do understand.  I'm not finding a "files on demand" box in my Onedrive settings.  I don't know if they took the option out in a later version, or if I just need to look somewhere else.

The laptop I'm using came with Onedrive pre-installed and it also replaces the file manager.  So, saving something to myPC>Documents>Arduino automatically saves it to Onedrive and has the location of C:\Users\dmike\OneDrive\Documents\Arduino.  That being said, I don't think the symbolic fix would work for me, either.  I don't know what else to do here.

facchinm, 2018-04-12:
go1.10 should already contain the fix (https://github.com/golang/go/commit/e83601b4356f92f2f4d05f302d5654754ff05a6d), I'm going to rebuild the builder with this version and merge it in the beta channel later today. I'll post here as soon as the builds are ready to let you test if this problem is finally gone.

facchinm, 2018-04-12:
I just pushed the builder rebuilt with go 1.10.1 to [IDE beta channel](https://www.arduino.cc/en/Main/Software#beta) and [here](https://github.com/arduino/arduino-builder/pull/250#issuecomment-380714621), if anyone could test it on a Windows PC with OneDrive Document folder it would be great
@species5618  @dmikedaniel  @jlvandusen @ssylvan @matthijskooijman

ngmacha, 2018-04-12:
Thank you for trying to fix this, @facchinm! Unfortunately it still doesn't work. I've just tried (Arduino-PR-beta1.9-BUILD-41), and still receiving error "Unable to find..." with full path to the sketch file. Same error when pointing directly to the OneDrive folder or the symlink. BTW, the symlink workaround initially worked, but stopped after March updates to Win10...

species5618, 2018-04-12:
The only reliable workaround itvtonplace your code
Libraries etc

Outside of onedrive 

But you can create a symbolic link into onedrive so it will get synced ok

matthijskooijman, 2018-04-12:
@ngmacha, could you show the literal error, preferably including some lines of output before the error?

ngmacha, 2018-04-12:
Here is the full output:

> Arduino: 1.9.0-beta (Windows 10), Board: "Arduino/Genuino Uno"
>
> D:\Apps\arduino-PR-beta1.9-BUILD-41\arduino-builder -dump-prefs -logger=machine -hardware D:\Apps\arduino-PR-beta1.9-BUILD-41\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -tools D:\Apps\arduino-PR-beta1.9-BUILD-41\tools-builder -tools D:\Apps\arduino-PR-beta1.9-BUILD-41\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries D:\Apps\arduino-PR-beta1.9-BUILD-41\libraries -libraries C:\Users\HaMac\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10900 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_54129 -warnings=all -build-cache C:\Users\HaMac\AppData\Local\Temp\arduino_cache_344812 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Arduino\BlinkTwoLEDs\BlinkTwoLEDs.ino
>
> Unable to find C:\Users\HaMac\OneDrive\Documents\Arduino\BlinkTwoLEDs\BlinkTwoLEDs.ino in C:\Users\HaMac\OneDrive\Documents\Arduino\BlinkTwoLEDs
>
> Error compiling for board Arduino/Genuino Uno.

matthijskooijman, 2018-04-12:
Ok, so the error is different from previous versions, which mentioned `readlink`, `open` or `GetFileAttributesEx` (unless the error messages were changed in this version of `arduino-builder`, @facchinm ?)

facchinm, 2018-04-12:
Nope, the error message was not changed; I think the Go fix brings in some subtle differences in functions like `isDir()` (in case of OneDrive mountpoints).

ngmacha, 2018-04-12:
If this may help, I rerun with debug-level 20, here is the full output:

>d:\Apps\arduino-PR-beta1.9-BUILD-41>D:\Apps\arduino-PR-beta1.9-BUILD-41\arduino-builder -dump-prefs -logger=machine -hardware D:\Apps\arduino-PR-beta1.9-BUILD-41\hardware -hardware C:\Users\HaMac\AppData\Local\Arduino15\packages -tools D:\Apps\arduino-PR-beta1.9-BUILD-41\tools-builder -tools D:\Apps\arduino-PR-beta1.9-BUILD-41\hardware\tools\avr -tools C:\Users\HaMac\AppData\Local\Arduino15\packages -built-in-libraries D:\Apps\arduino-PR-beta1.9-BUILD-41\libraries -libraries C:\Users\HaMac\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10900 -build-path C:\Users\HaMac\AppData\Local\Temp\arduino_build_54129 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.arduinoOTA.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avr-gcc.path=C:\Users\HaMac\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\HaMac\OneDrive\Documents\Arduino\BlinkTwoLEDs\BlinkTwoLEDs.ino -debug-level 20
===info ||| Progress {0} ||| [0.00]
===info ||| Progress {0} ||| [33.33]
===info ||| Progress {0} ||| [66.67]
===info ||| Progress {0} ||| [69.05]
===info ||| Progress {0} ||| [71.43]
===info ||| Progress {0} ||| [73.81]
===info ||| Progress {0} ||| [76.19]
===info ||| Progress {0} ||| [78.57]
===info ||| Progress {0} ||| [80.95]
===info ||| Using board '{0}' from platform in folder: {1} ||| [uno C%3A%5CUsers%5CHaMac%5CAppData%5CLocal%5CArduino15%5Cpackages%5Carduino%5Chardware%5Cavr%5C1.6.21]
===info ||| Using core '{0}' from platform in folder: {1} ||| [arduino C%3A%5CUsers%5CHaMac%5CAppData%5CLocal%5CArduino15%5Cpackages%5Carduino%5Chardware%5Cavr%5C1.6.21]
===info ||| Progress {0} ||| [83.33]
===info ||| Progress {0} ||| [85.71]
===info ||| Progress {0} ||| [88.10]
===error ||| Unable to find {0} in {1} ||| [C%3A%5CUsers%5CHaMac%5COneDrive%5CDocuments%5CArduino%5CBlinkTwoLEDs%5CBlinkTwoLEDs.ino C%3A%5CUsers%5CHaMac%5COneDrive%5CDocuments%5CArduino%5CBlinkTwoLEDs]

species5618, 2018-04-12:
from many hours lost trying different things, it is a one drive problem
the way one drive monitors files changes seems to block the java based Arduino software from opening files that are clearly in the file system

matthijskooijman, 2018-04-12:
@species5618, the problem seems to be that the go runtime libraries do not understand the way Onedrive encodes its files specially in the filesystem. This has partially been fixed in the go language, but presumably not entirely. I do not believe your comments are really helpful towards a fix in this area.

dmikedaniel, 2018-04-12:
I'm so glad there are people out there that are smarter than me and know what this stuff is, lol.

Can I ask what Go is and how to use it?

matthijskooijman, 2018-04-12:
Go is the programming language used to program `arduino-builder`. I'm not sure this is the right place for these questions, though... :-)

facchinm, 2018-04-12:
I'm trying to reproduce the issue on several machines with no luck. The "file not found" is surely related with https://github.com/arduino/arduino-builder/blob/master/gohasissues/go_has_issues.go#L41 , but it's very difficult to debug it without any hint :slightly_smiling_face:

dmikedaniel, 2018-04-12:
If it helps, the Arduino IDE worked fine on my Windows 10 machine with Onedrive until the April 10 update.

https://support.microsoft.com/en-us/help/4093112/windows-10-update-kb4093112

I only ask about Go because it seems like it might help if I knew how to use the fix (assuming the fix uses Go).  I saw one of the links by @facchinm above mention replacing the arduino-builder, but I don't exactly know where to find it.  

Also, I'm using the Microsoft Store Arduino Desktop App rather than downloading the program and installing it (which might be why I can't find the arduino-builder).  I'm not sure how much difference that makes, but when I tried to download the program, I kept getting redirected to the Microsoft Store so I just went with it.

Sorry if I'm cluttering the thread with unrelated basics that I probably should know.

per1234, 2018-04-13:
We're seeing a significant increase of reports of this issue on the Arduino forum over the last couple of days. It seems like maybe something happened with that last Windows update that made the issue more widespread (as supported by @dmikedaniel's comment).

facchinm, 2018-04-13:
@per1234 thanks for cross referencing the forum threads. I'm doing some other tests today with the very latest Windows build (and OneDrive) and try to figure out a solution.

facchinm, 2018-04-13:
Still no luck, I've been testing a lot of combinations (IDE and/or sketchbook inside the oneDrive folder, file-on-demand on and off, beta/1.8.5/win store version, synced Documents folder) on WIn10  17127, OneDrive 2018 (build 18.065.0329.0002) and every version simply works as expected.
At this point a teamviewer session with someone exposing the problem looks like the only way to go :slightly_smiling_face:

ngmacha, 2018-04-13:
I think you are running Windows Insiders version?..

facchinm, 2018-04-13:
@ngmacha yes, it's a Windows insider build. Can you paste her your configuration (Windows build and "flavour" (home/pro/x86_64 etc..) and OneCloud version? Thanks

ngmacha, 2018-04-13:
I tryin all this on two PCs (and all this hassle is to have all Arduino sketches synchronized between them):
1. Win10 Pro, Version 1709, OS Build 16299.309; OneDrive version 2018 Build 18.044.0301.0006;
2. Win10 Home, Version 1709, OS Build 16299.371, OneDrive version 2018 Build18.044.0301.0006;

ngmacha, 2018-04-13:
FYI, @facchinm, just tried on a test machine running Insider's build, everything works fine!! It is a Win10 Pro, Version 1803, OS Build 17133.73, OneDrive version 2018 Build 18.065.0329.0002

facchinm, 2018-04-13:
I've finally been able to reproduce it in Build 16299.309, fix is ongoing :slightly_smiling_face:

facchinm, 2018-04-13:
Shame on me, I misinterpreted the `git describe` output... The patch was merged after 1.10.1 and thus it will land in mainline with go 1.11 .
Anyway, compiling from master branch works, so I'm pasting here the exe if someone wants to test it
[arduino-builder-1.9.0.zip](https://github.com/arduino/arduino-builder/files/1907852/arduino-builder-1.9.0.zip)
[arduino-builder-1.8.5.zip](https://github.com/arduino/arduino-builder/files/1907853/arduino-builder-1.8.5.zip)

ngmacha, 2018-04-13:
Great job, @facchinm: just tested with 1.8.5, works as it should be!! Thank you very much!!

LRagji, 2018-04-15:
Tested arduino-builder-for-1.8.5.zip it works awesome @facchinm

NiordSailor, 2018-04-15:
Thank you, thank you, thank you. It works with my Win10Pro and OneDrive also.
Ver 1.8.5.zip Just replaced the arduino-builder.exe file with the new one.

xescure, 2018-04-15:
@facchinm Thank you, works for me too. When should we expect it to go live?

jlvandusen, 2018-04-16:
Windows 10 Pro
Build 16299.371

I am still getting the same thing when I load my onedrive hosted INO sketch… see below
readlink C:\Users\jlvan\OneDrive\Documents\Arduino\Sketches\Shadowv.2\Shadowv.2.ino: The system cannot find the file specified.
Error compiling for board Arduino Mega ADK.

elatrium, 2018-04-16:
(finally) working on W10  x64  home version with OneDrive using Portable 1.8.5

arielnh56, 2018-04-16:
Fixed my win10 Home problem on IDE 1.8.5. I had to remove the Windows Store install and reinstall fram the Windows Installer so as to figure out where to put the binary, but that's just my lack of Windows foo.

many thanks

facchinm, 2018-04-16:
Go 1.11 will be published at the end of July, so it's quite a long time. I have to discuss with @cmaglie whether to publish this binary (compiled with a non-stable version of go) or to leave it in the beta channel (which is, in its own nature, prone to hosting non stable software).

xescure, 2018-04-16:
@arielnh56 The Microsoft Store version is located at `C:\Program Files\WindowsApps\<the folder containing the word Arduino>`, but I've ran into permission issues so I ended up uninstalling it too. Let me know if anyone was successful fixing it for the Microsoft Store version.

matthijskooijman, 2018-04-16:
@facchinm, I would tend to not release things compiled with a git version of go and either wait for 1.11, or backport just this one fix to go 1.10 if possible (but that might be a boatload of extra work).

thundo, 2018-04-17:
Fixed for me too on 1.8.5.

Win 10 Pro 10.0.16299

Thanks

facchinm, 2018-04-17:
I just pushed the "fixed" builder on the [Beta channel](https://www.arduino.cc/download_handler.php?f=/ide_beta/arduino-1.9.0-beta-windows.zip) , so there is finally a way to use a self contained version of the IDE if you encounter this problem.
@per1234 do you believe it is ok to instruct people asking on the forum to try the beta?

per1234, 2018-04-17:
I certainly do mention the beta or hourly builds whenever it might solve a problem someone reports on the forum, but with the warning that these builds are primarily intended for beta testing and so there is not so much a guarantee of stability as they would have with the production IDE releases.

I've been more hesitant to recommend the Beta due to the known [unresolved issues](https://github.com/arduino/arduino-preprocessor/issues) with arduino-preprocessor. The FastLED incompatibility is my primary concern since that's such a popular library. I believe the Arduino Web Editor is using arduino-preprocessor though and I haven't noticed many reports of problems there lately so maybe it's not such an issue as I think.

In this case there are multiple solutions to the issue so if they encounter a problem with the beta IDE they have other options.

RoyAnderson, 2018-04-25:
builder update for 1.8.5 fixed my OneDrive problem. Thanks.

Prehistoricman, 2018-04-26:
Just got this error and builder 1.8.5 fixed it.
What's the issue with pushing the new builder to the master branch?

facchinm, 2018-04-27:
@Prehistoricman the issue is that you need the master branch of `go` to compile it, and it could contain any sort of bug/regression we won't discover until June.
Since the IDE could be released tomorrow if we want to, merging it in master would make it available in binary form for a long time.

liamkennedy, 2018-05-10:
I got this error with the 1.8.5 App store version.  So... not sure what may be different with my setup as there are multiple posts above (13+days ago) saying that fixed this issue.  

I am running on a Surface Book with Windows 10.0.16299

To get this working I had to install the 1.9.0 BETA (via the zip).

Just thought others who come here should know.  

Thanks!

arteegee, 2018-05-11:
I am probably the most ignorant guy to be in this conversation, so I have nothing to contribute, except to add to the list of complainers. I don't understand you experts' tech language and complicated "fixes". At least I can see there is a legitimate problem and there is some kind of action goin' 'round.  I am anxiously waiting for a final, conclusive bottom line fix. Thanks for the explanation, up to now.  I am at least satisfied that I have not gone bonkers. Thanks to all you contributors, and I'll stay tuned.

xescure, 2018-06-19:
@facchinm just downloaded the hourly build, I'm getting no errors. Thanks for fixing it. When should we expect it to come to the Microsoft Store version?

PaulStoffregen, 2018-07-30:
@facchinm - Any idea of a time frame for 1.8.6?  Would be really nice to get this OneDrive fix pushed out to Windows users!

xescure, 2018-08-21:
> (this will happen in max 2 weeks from now)

Any update on this? When will the 1.8.6 be released? @facchinm

facchinm, 2018-08-21:
@xescure we had *some* delays due to vacations and other priorities, but I can assure you that it will be published during this week

[2 more replies]

closed 2018-08-28 by facchinm