arduino/arduino-builder #255: tests fail with board 'my_avr_platform' with 32-bit Intel host

andreasbaumann, 2017-11-23:
go test arduino.cc/...

fails on 32-bit Intel with:

```
Warning: Board my_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
Warning: Board my_symlinked_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set t
o: AVR_MYMEGA
Warning: Board my_symlinked_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set t
o: AVR_MYMEGA
...
--- FAIL: TestBuilderEmptySketch (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    helper.go:72
                ^M                      ^M                      builder_test.go:74
                ^M      Error:          Received unexpected error:
                ^M                      stat /tmp/test_build_path564566314/core/HardwareSerial.cpp.o: no such file or directory
--- FAIL: TestBuilderBridge (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    helper.go:72
                ^M                      ^M                      builder_test.go:99
                ^M      Error:          Received unexpected error:
                ^M                      stat /tmp/test_build_path582351489/core/HardwareSerial.cpp.o: no such file or directory
--- FAIL: TestBuilderSketchWithConfig (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    helper.go:72
                ^M                      ^M                      builder_test.go:126
                ^M      Error:          Received unexpected error:
                ^M                      stat /tmp/test_build_path162130412/core/HardwareSerial.cpp.o: no such file or directory
--- FAIL: TestBuilderBridgeTwice (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    helper.go:72
                ^M                      ^M                      builder_test.go:158
                ^M      Error:          Received unexpected error:
                ^M                      stat /tmp/test_build_path205896795/core/HardwareSerial.cpp.o: no such file or directory
--- FAIL: TestBuilderBridgeSAM (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    helper.go:72
                ^M                      ^M                      builder_test.go:186
                ^M      Error:          Received unexpected error:
                ^M                      stat /tmp/test_build_path209285118/core/syscalls_sam3.c.o: no such file or directory
--- FAIL: TestBuilderBridgeRedBearLab (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    helper.go:72
                ^M                      ^M                      builder_test.go:224
                ^M      Error:          Received unexpected error:
                ^M                      stat /tmp/test_build_path031066437/core/HardwareSerial.cpp.o: no such file or directory
--- FAIL: TestBuilderSketchNoFunctions (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    builder_test.go:250
                ^M      Error:          An error is expected but got nil.
--- FAIL: TestBuilderSketchBuildPathContainsUnusedPreviouslyCompiledLibrary (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    builder_test.go:313
                ^M      Error:          An error is expected but got nil.
--- FAIL: TestBuilderCacheCoreAFile (0.00s)
        assertions.go:239: ^M                          ^M       Error Trace:    builder_test.go:360
                ^M      Error:          Received unexpected error:
                ^M                      stat core_arduino_avr_uno_5a0232b855defa2754e82e47db0a2b8a.a: no such file or directory
...
Warning: Board my_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
Warning: Board my_symlinked_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
Build options changed, rebuilding all
FAIL
FAIL    arduino.cc/builder/test 87.396s
```

andreasbaumann, 2017-11-23:
version is 1.3.25

facchinm, 2017-11-27:
Hi @andreasbaumann ,
do you have user access rights to /tmp/ ? On our Linux32 CI machine (Debian7) all the tests pass just fine.

andreasbaumann, 2017-11-27:
I'm running in a 32-bit chroot on Archlinux (as root). And it's the only test failing, all others pass fine.

facchinm, 2017-11-27:
Well, that's strange. I'll test it and report back soon. Thanks!

andreasbaumann, 2017-11-27:
Well, if your 32-bit CI says "all ok", I'll also have to dig what's wrong in my build environment..

andreasbaumann, 2017-11-27:
I have a:

```
warning: ignoring symlink /build/arduino-builder/src/arduino-builder-1.3.25/src/arduino.cc/builder/test/user_hardware/my_symlinked_avr_platform
```
 when running the tests.

andreasbaumann, 2017-11-27:
typical memory usage in /tmp (tmpfs):

```
tmpfs             4056404     63196    3993208   2% /tmp
```

looks ok to me.

andreasbaumann, 2017-11-27:
I see some other warnings:

```
Warning: Board my_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
Warning: Board my_symlinked_avr_platform:avr:mymega doesn't define a 'build.board' preference. Auto-set to: AVR_MYMEGA
```
Maybe as a side-effect certaings files are missing afterwards?

andreasbaumann, 2017-11-27:
Found this bug (also Archlinux), related?

https://github.com/arduino/arduino-builder/issues/202

andreasbaumann, 2017-11-27:
I retested on Archlinux 64-bit, there is no problem there.. 

I would assume, the bulder doesn't generate any output for some tests and then the test code bails
out with an assertion.

Any way I can see the builder command being executed while in the test case?
go -v (verbose) didn't produce more output..