arduino/arduino-builder #256: Precompiled libs feature doesn't account for floating point mcu variants

kgoveas, 2017-11-30:
Hi

The precompiled libs fix for the arduino builder doesn't account for the floating point unit variants of the mcu. Example: {build.mcu} = cortex-m4 but there are 2 variants of the MCU. With floating point and without floating point. 

Unfortunately i have no input for how to know which mcu variant is in use. But the ldflags would be found in cortex-m4 or cortex-m4f.

facchinm, 2017-12-01:
Hi @kgoveas ,
you are right, I didn't think about this case.
What about using the multilib approach from this table (https://stackoverflow.com/a/25728024) ?

kgoveas, 2017-12-01:
Hi @facchinm ,

Do you mean like a folder structure like cortex-m4, cortex-m4-sfp, cortex-m4-hfp? If so, what references would you use from the platform.txt file apart from the {build.mcu}?

facchinm, 2017-12-04:
Probably {build.mfloat-abi} is the only way to extract that information but it must be declared in board.txt and not directly on the commandline... It's a bit awkward but the most effective thing to do could be creating two different libraries (one for hardfloat variant) and including them selectively (maybe with `#error` pragmas to make it clear for the user that the library he specified is not compatible with the selected target)

kgoveas, 2017-12-04:
How about using {build.arch} instead? It would create duplicates in case of archs that share the same MCU but it would also synchronize with the warning that would appear in the Arduino IDE.

facchinm, 2018-03-15:
Ok, I had some time to take a look at the issue; the correct approach is probably the one adopted by STM32 core, which specifies all the information in `build.mcu`(https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt#L133). This will totally work, although the folder name will be horrible. 
Extracting the various parameters from the generated commandline would make things much more difficult. What to you think about it?

kgoveas, 2018-03-18:
Hi @facchinm ,

Yes, i guess this would work. I had come across this while testing on the STM32F411RE board.

closed 2020-02-11 by cmaglie