arduino/arduino-builder #258: fatal error: memory: No such file or directory

jankalfus, 2017-12-24:
I am getting this error in Arduino 1.8.5 when including FS.h for ESP32 in a .cpp module.
I have a file structure like this:
```
|-- main.ino (contains #include "files.hpp")
|-- files.cpp (contains #include "files.hpp")
|-- files.hpp
```

Content of files.hpp:
```
#ifndef FILES_H
#define FILES_H

#include "FS.h"

void appendFile(fs::FS &fs, const char *path, const char *message);
void readFile(fs::FS &fs, const char *path);

#endif
```

The error message:
```
Compiling sketch...
"C:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu99 -Os -g3 -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wpointer-arith  -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -MMD -c -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32\variants\esp32" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src" "C:\Users\janka\AppData\Local\Temp\arduino_build_542757\sketch\files.c" -o "C:\Users\janka\AppData\Local\Temp\arduino_build_542757\sketch\files.c.o"
In file included from C:\Users\janka\AppData\Local\Temp\arduino_build_542757\sketch\files.h:4:0,

                 from C:\Users\janka\AppData\Local\Temp\arduino_build_542757\sketch\files.c:1:

C:\Users\janka\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src/FS.h:24:18: fatal error: memory: No such file or directory

compilation terminated.
```

This might be related to #68. Is the preprocessor broken again or I'm doing something wrong? :(

cmaglie, 2017-12-24:
From the verbose command line it seems that in your sketch you have a file named `files.c`:

"C:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu99 -Os -g3 -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wpointer-arith  -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -MMD -c -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32\variants\esp32" "-IC:\Users\janka\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src" **"C:\Users\janka\AppData\Local\Temp\arduino_build_542757\sketch\files.c"** -o "C:\Users\janka\AppData\Local\Temp\arduino_build_542757\sketch\files.c.o"

The <memory> include is a C++-only feature, so you should rename `files.c` to `files.cpp` in order for the IDE to use g++ to compile.

jankalfus, 2017-12-24:
I don't, the file name is `files.cpp`. In the sketch file, I only have `#include "files.hpp"`. I have no idea why it tries to compile `files.c`.

cmaglie, 2017-12-24:
maybe it's a stray file, did you rename it? try to close and reopen the IDE

jankalfus, 2017-12-24:
Restarting the IDE helped! Thank you so much!

It's weird I had to restart it though. Is there some logic behind that? Do I have to restart it every time I make a change? Is there a clean build command? I don't seem to be able to find one.

closed 2017-12-24 by jankalfus