arduino/arduino-builder #259: Library path issues with old compilers

alvieboy, 2017-12-25:
I am currently using an old GCC compiler (gcc 3.4.2) and I am having issues with library detection.

Under "findIncludeForOldCompilers" the current code searches for "fatal error" which is not output by this compiler. Example for SPI library include:

gcc3.4.2 (not working):
`/home/alvieboy/sketchbook/tft/tft.ino:1:17: SPI.h: No such file or directory`
gcc4.8.2 (different platform, working)
`/home/alvieboy/sketchbook/tft/tft.ino:1:17: fatal error: SPI.h: No such file or directory`

This causes my system not to include any library due to match failure. The folllowing code fixes it, but is awkward, and may not work with localization enabled.

I cannot upgrade compiler as of now. Perhaps there's a better way to filter out the includes ?
```diff
+++ b/src/arduino.cc/builder/includes_finder_with_regexp.go
@@ -62,6 +62,9 @@ func findIncludeForOldCompilers(source string) string {
                        if strings.Contains(splittedLine[i], "fatal error") {
                                return strings.TrimSpace(splittedLine[i+1])
                        }`
+                       if strings.Contains(splittedLine[i], "No such file") {
+                               return strings.TrimSpace(splittedLine[i-1])
+                       }
                }
        }
        return ""
```
arduino-builder: git 03d0da61f80d899c98e57037f4efffe3410a7d53

matthijskooijman, 2017-12-26:
Hm, so apparently this is an even older compiler :-)

Your proposed fix isn't entirely elegant, but since it is just a workaround for older compilers, I think that should be more acceptable.

As for the locale issue, that already seems to apply with the `fatal error` matching, see #243 for a proposed fix.

@cmaglie, I haven't looked at arduino-preprocessor yet, but is that also going to handle preprocessing and include detection? If so, are these issue going to be irrelevant?

alvieboy, 2017-12-26:
Thanks for your feedback Matthijs.
I do have another option, which is to sligtly modify the compiler to output the "fatal error" string. I may do that just to be compatible with 1.8.5 Arduino release.
Let me know if you want me to do a pull request for this workaround (I assume you will probably want to hear Cristian comments prior to apply this change)
Best,
Alvie

matthijskooijman, 2017-12-26:
> I assume you will probably want to hear Cristian comments prior to apply this change)

Let's wait for that, I can't apply PR's anyway, I just review them  :-)

cmaglie, 2017-12-26:
> @cmaglie, I haven't looked at arduino-preprocessor yet, but is that also going to handle preprocessing and include detection? If so, are these issue going to be irrelevant?

It does only the preprocessing -> adding forward prototypes where missing
(in other words it is going to replace ctags)