arduino/arduino-builder #261: Compiling non target ino file

Amertz08, 2018-01-18:
It appears that builder is parsing a non target ino file (`MostBasicFollower.ino`) in the same directory as my desired file (`ultra.ino`). I ran the following command and got the follow output

```bash
Adams-MacBook-Pro:sensors adammertz$ make ultra
arduino-builder -hardware /Applications/Arduino.app/Contents/Java/hardware \
	-tools /Applications/Arduino.app/Contents/Java/hardware/tools \
	-fqbn arduino:avr:uno \
	-libraries /Applications/Arduino.app/Contents/Java/libraries \
	-warnings ALL \
	/Users/adammertz/development/Robot/arduino/sensors/ultra.ino
/Users/adammertz/development/Robot/arduino/sensors/MostBasicFollower.ino:1:34: fatal error: Adafruit_MotorShield.h: No such file or directory
 #include <Adafruit_MotorShield.h>
                                  ^
compilation terminated.
exit status 1
make: *** [ultra] Error 1
```

I knew the library wasn't installed as I had a fresh install of the IDE and had no 3rd party libraries installed yet (I'd later realize I also had the library flag pointing at the wrong directory). I installed the library and reran the command (pointing library at the correct destination) and had a similar error appear again.

```bash
Adams-MacBook-Pro:sensors adammertz$ make ultra
arduino-builder -hardware /Applications/Arduino.app/Contents/Java/hardware \
	-tools /Applications/Arduino.app/Contents/Java/hardware/tools \
	-fqbn arduino:avr:uno \
	-libraries /Users/adammertz/Documents/Arduino/libraries \
	-warnings all \
	/Users/adammertz/development/Robot/arduino/sensors/ultra.ino
Build options changed, rebuilding all
/Users/adammertz/development/Robot/arduino/sensors/MostBasicFollower.ino:50:23: fatal error: sensorbar.h: No such file or directory
 #include "sensorbar.h"
                       ^
compilation terminated.
exit status 1
make: *** [ultra] Error 1
```

I get that not having the libraries installed would throw an error on compilation but I am not sure why `MostBasicFollower.ino` is being parsed in the first place.

matthijskooijman, 2018-01-19:
This is by design: A sketch is composed of all .ino files in the sketch directory, which allows splitting up code into multiple files. I think there is no bug here, so I'm closing this issue. Feel free to comment and/or reopen if you think differently.

closed 2018-01-19 by matthijskooijman

Amertz08, 2018-01-19:
Ok sounds good.