arduino/arduino-builder #263: avr-g++: error: cannot access device-specs for 'atmega328p' expected at 'device-specs/specs-atmega328p'

kenneth558, 2018-02-05:
Ubuntu 16.04 with KDE installed but this error is via ssh at the CLI.  Newly installed using the latest golang, arduino-builder and IDE, all installed from respective tar.gz-type files (none from Ubuntu repository).  I got the error with and without running the IDE for it to make the ~/.arduino15 directory on its first compile+download.  

(Note that one error along the way was

> fork/exec /bin/avr-g++: no such file or directory

for which I made the following link:

> ln /home/homeowner/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++ /bin

Maybe that was the incorrect way to handle that?????)

facchinm, 2018-02-06:
Hi @kenneth558 ,
the builder can also work in a fully standalone way when all the paths are configured correctly. The first command is missing `tools-builder` folder which contains `ctags`, needed to forward-declare missing prototypes. 

Furthermore, `/bin/avr-g++` error means that `{runtime.avr-gcc.path}` could not be expanded by the builder since it didn't find any matching tool. On a normal installation (if you only need avr support) tools live in ` $arduinoBaseFolder/hardware/tools/avr` so you need to add that path to `-tools` to let the builder find the compiler. No symlink needed :wink: 

Please close the issue if this answers your doubts, thanks!

kenneth558, 2018-02-06:
I'll close the issue b/c I'm confident I can enact a workaround for my own needs.  The puzzling fact that there are multiple versions of the IDE 1.8.5 (one with an the other without arduino-builder) hopefully will not totally block my project's progress further, though it does indicate teamwork issues.

As I closed this, I was in the process of looking at the contrasting outputs of the two IDE tar files as shown by running the following command on both of them:

> tar -tf Downloads/arduino-1.8.5-linux64.tar.xz|grep builder
and 
> tar -tf Downloads/Arduino-1.8.5.tar.xz|grep builder

The differences between the two zip files, which I don't include here, should be made known to the development team members involved.

closed 2018-02-06 by kenneth558

facchinm, 2018-02-07:
`Arduino-1.8.5.tar.xz` is not an official build. All the official ones are listed on https://www.arduino.cc/en/Main/Software and [these](https://downloads.arduino.cc/arduino-1.8.5.md5sum.txt) are the checksums.
Any other distribution is to be considered unofficial and totally unsupported. Thanks

kenneth558, 2018-02-07:
Very nice to know!  Thank you and Lord bless!!!