arduino/arduino-builder #274: arduino-builder doesn't refer to libraries by their library.properties name value

per1234, 2018-05-17:
Using Arduino IDE 1.9.0 Beta build 58 and Arduino IDE 1.8.6 Hourly Build 2018-05-15 10:41

1. File > Preferences > Show verbose output during: > compilation (check) > OK
1. Tools > Board > Arduino/Genuino Zero
2. Compile the following sketch:
    ```cpp
    #include <ArduinoRobotMotorBoard.h>
    void setup() {}
    void loop() {}
    ```
The console displays the following warning:
```
WARNING: library Robot_Motor claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (samd) architecture(s).
```
and
```
Compiling library "Robot_Motor"
```
and
```
Using library Robot_Motor at version 1.0.3 in folder: C:\Program Files (x86)\ArduinoIDE\arduino-PR-beta1.9-BUILD-58\libraries\Robot_Motor 
```
But the `name` value defined in library.properties for this library is `Robot Motor`. The "Robot_Motor" in the warning is the library folder name.

I realize the difference between the two is not very dramatic in this demonstration. I chose this library because Robot Motor library is bundled with the Arduino IDE so it provides a minimal demonstration of the issue. The issue is more noticeable with other libraries where the folder name differs more significantly from the `name` value.

The Arduino IDE uses the `name` value whenever referring to a library (including the invalid category warning shown in the console) so this causes an inconsistent user experience.

Two of these messages are already using information from library.properties so I can't imagine it would be too difficult to use the `name` value as well (falling back on the folder name when a library.properties file is absent).

facchinm, 2018-05-17:
@per1234 thanks for noticing, the name being printed is the one used internally (no whitespace and some other small cosmetic stuff based on the containing folder).
Could something like this https://github.com/facchinm/arduino-builder/commit/c20627eeb7e5dbd7a1366c69b5466d839aa9681d work? (of course, in all the mayor places and not only on the "incompatible" string)

matthijskooijman, 2018-05-17:
Sounds good to me. Since we're already printing a list of used libraries with their full paths at the end of the compilation, it doesn't seem relevant to use the folder name in all debug output (you can map the fancy names to folders using that list printed at the end).

Would it make sense to have a method to retrieve the fancy name (and default to the the regular name if unset), or perhaps set `RealName` to `Name` when loading if the former is unset to prevent duplicating this defaulting in a bunch of places?

e-ika, 2018-09-06:
Thanks for noticing per1234