arduino/arduino-builder #278: arduino-builder destroys template code

feklee, 2018-06-08:
Compilation of this valid C++11 code fails:

    template <int x>
    class A {
      // use x...
    };
    
    template <int x>
    void f(A<x> a) {
      // do something...
    }
    
    void setup() {}
    
    void loop() {}

Log from the IDE:

    C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware -hardware C:\Users\Felix\AppData\Local\Arduino15\packages -tools C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\tools-builder -tools C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -tools C:\Users\Felix\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\libraries -libraries C:\Users\Felix\Documents\Arduino\libraries -fqbn=arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10806 -build-path C:\Users\Felix\AppData\Local\Temp\arduino_build_579189 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -verbose D:\Dropbox\Temp Work\drafts\Arduino\Arduino.ino
    C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware -hardware C:\Users\Felix\AppData\Local\Arduino15\packages -tools C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\tools-builder -tools C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -tools C:\Users\Felix\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\libraries -libraries C:\Users\Felix\Documents\Arduino\libraries -fqbn=arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10806 -build-path C:\Users\Felix\AppData\Local\Temp\arduino_build_579189 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -verbose D:\Dropbox\Temp Work\drafts\Arduino\Arduino.ino
    Using board 'pro' from platform in folder: C:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20
    Using core 'arduino' from platform in folder: C:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20
    Detecting libraries used...
    "C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "C:\Users\Felix\AppData\Local\Temp\arduino_build_579189\sketch\Arduino.ino.cpp" -o "nul"
    Generating function prototypes...
    "C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "C:\Users\Felix\AppData\Local\Temp\arduino_build_579189\sketch\Arduino.ino.cpp" -o "C:\Users\Felix\AppData\Local\Temp\arduino_build_579189\preproc\ctags_target_for_gcc_minus_e.cpp"
    "C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Felix\AppData\Local\Temp\arduino_build_579189\preproc\ctags_target_for_gcc_minus_e.cpp"
    Compiling sketch...
    "C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR   "-IC:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino" "-IC:\Users\Felix\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" "C:\Users\Felix\AppData\Local\Temp\arduino_build_579189\sketch\Arduino.ino.cpp" -o "C:\Users\Felix\AppData\Local\Temp\arduino_build_579189\sketch\Arduino.ino.cpp.o"
    Arduino:7:10: error: 'x' was not declared in this scope
    
     void f(A<x> a) {
    
              ^
    
    Arduino:7:11: error: template argument 1 is invalid
    
     void f(A<x> a) {
    
               ^
    
    D:\Dropbox\Temp Work\drafts\Arduino\Arduino.ino:7:13: warning: unused parameter 'a' [-Wunused-parameter]
    
     void f(A<x> a) {
    
                 ^
    
    exit status 1
    'x' was not declared in this scope

The problem is the second command, `arduino-builder -compile`, which creates a garbled C++ file:

    #include <Arduino.h>
    #line 1 "D:\\Dropbox\\Temp Work\\drafts\\Arduino\\Arduino.ino"
    #line 1 "D:\\Dropbox\\Temp Work\\drafts\\Arduino\\Arduino.ino"
    template <int x>
    class A {
      // use x...
    };
    
    template <int x>
    #line 7 "D:\\Dropbox\\Temp Work\\drafts\\Arduino\\Arduino.ino"
    void f(A<x> a) ;
    #line 11 "D:\\Dropbox\\Temp Work\\drafts\\Arduino\\Arduino.ino"
    void setup();
    #line 13 "D:\\Dropbox\\Temp Work\\drafts\\Arduino\\Arduino.ino"
    void loop();
    #line 7 "D:\\Dropbox\\Temp Work\\drafts\\Arduino\\Arduino.ino"
    void f(A<x> a) {
      // do something...
    }
    
    void setup() {}
    
    void loop() {}

Note the missing `template <int x>` in front of the function definition.

Tried with latest stable (1.8.5 / Windows 7x64), and with the latest hourly (2018-06-08 17:32 UTC+2):

    C:\Users\Felix>C:\Users\Felix\Desktop\arduino-nightly-windows\arduino-nightly\arduino-builder -version
    Arduino Builder 1.3.25
    Copyright (C) 2015 Arduino LLC and contributors
    See https://www.arduino.cc/ and https://github.com/arduino/arduino-builder/graphs/contributors
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

per1234, 2018-06-08:
Try the beta build:
https://www.arduino.cc/en/Main/Software#beta

I believe the fix was from https://github.com/arduino/arduino-builder/pull/182

feklee, 2018-06-09:
@per1234 Thanks, with the beta the bug is gone.

Interestingly when testing with the hourly, as described, I did experience the bug. I hope it does not resurface in a future version.

facchinm, 2018-06-11:
That fix has been merged in mainline builder too, so in a few days it will appear is hourly (without `arduino-preprocessor` but with all the other fixes).