arduino/arduino-builder #280: Can't build Arduino-builder

fracturedsoftware, 2018-06-13:
Using Mac OS High Sierra. Downloaded and installed go language. Followed the instructions on building Arduino-builder:
go get github.com/go-errors/errors
go get github.com/stretchr/testify
go get github.com/jstemmer/go-junit-report
go get github.com/arduino/go-properties-map
go get github.com/arduino/go-timeutils
go get github.com/arduino/arduino-builder

These all worked fine. However the last instruction to actually build the tool fails

go build github.com/arduino/arduino-builder/arduino-builder

go/src/github.com/arduino/arduino-builder/grpc/rpc.go:15:2: cannot find package "github.com/fsnotify/fsnotify" in any of:
	/usr/local/go/src/github.com/fsnotify/fsnotify (from $GOROOT)
	/Users/peter/go/src/github.com/fsnotify/fsnotify (from $GOPATH)
go/src/github.com/arduino/arduino-builder/grpc/proto/builder.pb.go:17:8: cannot find package "github.com/golang/protobuf/proto" in any of:
	/usr/local/go/src/github.com/golang/protobuf/proto (from $GOROOT)
	/Users/peter/go/src/github.com/golang/protobuf/proto (from $GOPATH)
go/src/github.com/arduino/arduino-builder/grpc/proto/builder.pb.go:22:2: cannot find package "golang.org/x/net/context" in any of:
	/usr/local/go/src/golang.org/x/net/context (from $GOROOT)
	/Users/peter/go/src/golang.org/x/net/context (from $GOPATH)
go/src/github.com/arduino/arduino-builder/grpc/proto/builder.pb.go:23:2: cannot find package "google.golang.org/grpc" in any of:
	/usr/local/go/src/google.golang.org/grpc (from $GOROOT)
	/Users/peter/go/src/google.golang.org/grpc (from $GOPATH)

regards,
Peter

facchinm, 2018-06-13:
Hi @fracturedsoftware ,
there are a couple of dependencies added recently (the readme has not been updated yet).
The new dependencies list is
```
go get github.com/go-errors/errors
go get github.com/stretchr/testify
go get github.com/jstemmer/go-junit-report
go get -u github.com/arduino/go-properties-map
go get -u github.com/arduino/go-timeutils
go get google.golang.org/grpc
go get github.com/golang/protobuf/proto
go get golang.org/x/net/context
go get github.com/fsnotify/fsnotify
```

I'll update the README accordingly.

closed 2018-06-13 by facchinm

fracturedsoftware, 2018-06-13:
Hi Martino,

Thank you for this. I added the dependencies and ran the tests and all is fine.

Kind regards,

Peter