arduino/arduino-builder #281: How do i add debug symbols?

ChriD, 2018-06-21:
Hi,

How can i add debug symbols to the elf file?

facchinm, 2018-06-25:
Hi @ChriD ,
normally you need to add `-g` flag to a couple of places in the `platform.txt` for the core you are targeting.
For example, using `avr` core, debug switch is already there (see https://github.com/arduino/ArduinoCore-avr/blob/master/platform.txt#L23). The relevant recipes are `compiler.c.flags`, `compiler.c.elf.flags` and `compiler.cpp.flags` . 
Please consider that using LTO mangles the debug symbols considerably, so you may been to disable it before trying to attach a debugger (by removing all `-flto` occurrences in the same file and then recompiling the project)

closed 2018-06-25 by facchinm

ChriD, 2018-06-25:
Ah ok. Thank you very much!