arduino/arduino-builder #283: A further generalized idea of modifying UserSettings.txt sketch-wide

xcvista, 2018-08-13:
This is a generalized solution to #15 and #29. It adds one preprocessor macro to the sketch file, while achieving the same effects without introducing special files:

```C++
#pragma arduino
```

Whatever text comes after this preprocessor macro, to the end of the line, is treated as part of the UserSettings.txt. For example putting the followign lines in the sketch:

```C++
#pragma arduino compiler.c.extra_flags=-DNDEBUG
#pragma arduino compiler.cpp.extra_flags={compiler.c.extra_flags} -DTESTLIBRARY_BUFSIZE=100
```

is equivalent to adding the following lines to UserSettings.txt for the specific sketch:

```
compiler.c.extra_flags=-DNDEBUG
compiler.cpp.extra_flags={compiler.c.extra_flags} -DTESTLIBRARY_BUFSIZE=100
```

phd, 2018-08-25:
1. Will this pragma also affect build process for libraries if added to the sketch code?
If not, then it will not be quite useful as an extension to platform.txt.

2. This has the same security problems as #29. A malicious sketch can modify commands' paths without user's consent.

I have shared a pull request #282 which adds full platform.sketch.txt file support while providing an IDE configuration option for enabling it first.
Comments are welcome.

xcvista, 2018-08-25:
@phd 

1. Yes it will. This `#pragma` is handled no later than the library-introducing `#include` directives, and will affect the entire build process.
2. The paths of all executables being called by `arduino-build` can be limited with whitelisting: any subdirectory within the Arduino-related folders, and any subdirectory within the current sketch. It is up to the user not to download untrustworthy platforms.

phd, 2018-08-25:
1. Then it seems like a nice replacement for my pull request.
I could probably modify it soon to support this.

2. User may download sample sketches and they should not be allowed (by default) to mess with the system.
I think a configuration option in the Arduino IDE for this pragma would be a solution. Maybe even something like a "no/ask/yes" setting.

xcvista, 2018-08-25:
@phd The `_Pragma()` syntax need not to be supported right now, just `#pragma arduino`.