arduino/arduino-builder #290: Missing GPG signatures for 1.4.0 release

NicoHood, 2018-09-07:
At least not on Github. Where can I find a signed release?

cmaglie, 2018-09-10:
Sorry, I've added the missing signatures. https://github.com/arduino/arduino-builder/releases/tag/1.4.0

closed 2018-09-10 by cmaglie

NicoHood, 2018-09-10:
https://github.com/arduino/arduino-builder/releases/download/1.4.0/arduino-builder-1.4.0.tar.xz is still missing, only the signature, no zip was provided.

cmaglie, 2018-09-10:
oh wow, looks impossible but the script missed one file :-o
Now is fixed for good :-)

cmaglie, 2018-09-12:
becuase it's not released yet... it will be in a couple of hours, we are doing the last tests.

NicoHood, 2018-12-26:
Missing for 1.4.2 as well.

NicoHood, 2019-01-11:
Still missing..

cmaglie, 2019-01-11:
Is the source code enough? or do you need also binaries?

NicoHood, 2019-01-15:
I compile everything from source. But in general it is also a wise idea to sign precompiled binaries.