arduino/arduino-builder #292: Non-class function prototypes are sometimes inserted inside class definition.

oqibidipo, 2018-09-15:
A.k.a. "Hello class, I found by the road some functions that look a bit like what we alrady have. Let's throw them in."

Originally reported at http://forum.arduino.cc/index.php?topic=568492.0

Example code
```
class Foo {
    int blooper(int x) { return x+1; }
};

Foo foo;

void setup() {
  foo.setup();
}

void loop() {
  foo.loop();
}
```

With generated prototypes

```
#include <Arduino.h>
#line 1 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
#line 1 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
class Foo {
#line 7 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
void setup();
#line 11 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
void loop();
#line 2 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
    int blooper(int x) { return x+1; }
};

Foo foo;

void setup() {
  foo.setup();
}

void loop() {
  foo.loop();
}
```

This happens when the name of a function is the same as or a substring of a name of class method like `loop` is a substring of `blooper` in the example.

Tested with IDE version 1.8.7, Mac OS X 10.11.6.
OK with 1.8.5.

oqibidipo, 2018-09-15:
These produce different errors depending on the situation:
```
void setup() {
  foo.setup();
}

void loop() {
  foo.loop();
}
```
Correct result:
```
error: 'class Foo' has no member named 'setup'
error: 'class Foo' has no member named 'loop'
```
Buggy result:
```
error: 'void Foo::setup()' is private
error: 'void Foo::loop()' is private
```

facchinm, 2018-09-17:
The bug is caused by https://github.com/arduino/arduino-builder/commit/287a9c8e0f38263058e29db9425a3dbe74061f3c#diff-d9628c6c5e4ac5689a2cee322bf97ad5R73 ; I'm checking if there's a more clever way to check for matching signatures

oqibidipo, 2018-09-17:
Seems to do the right thing with my example.
```
#include <Arduino.h>
#line 1 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
#line 1 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
class Foo {
    int blooper(int x) { return x+1; }
};

Foo foo;

#line 7 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
void setup();
#line 11 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
void loop();
#line 7 "/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino"
void setup() {
  foo.setup();
}

void loop() {
  foo.loop();
}
```

Compilation fails with the 'correct' errors:
```
/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino: In function 'void setup()':
bug_proto_1:8:7: error: 'class Foo' has no member named 'setup'
   foo.setup();
       ^
/Users/rjl/Documents/Arduino/bug_proto_1/bug_proto_1.ino: In function 'void loop()':
bug_proto_1:12:7: error: 'class Foo' has no member named 'loop'
   foo.loop();
       ^
```

domenpk, 2018-09-17:
FWIW #294 fixes my issue.

closed 2018-10-16 by facchinm