arduino/arduino-builder #300: arduino-builder returned 1,073,741,855

Rashantha, 2018-10-08:
Hello,

We have been trying to get Arduino to run in our School.

We have tried installing via .exe
Zip folder with - portable file as per instructions.
Also we tried installing from the School Microsoft Education Store.

facchinm, 2018-10-09:
Hi @Rashantha ,
can you tell us a bit more about the problem you are facing? For example, which operating system are these PC running? Can you try compiling a sketch with verbose compilation on (from File/Preferences) and paste here the complete output? Thanks

Rashantha, 2018-10-09:
Same error if installed via .exe, zip file or MS Store.

Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\test.student\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10807 -build-path C:\Users\TEST~1.STU\AppData\Local\Temp\arduino_build_95736 -warnings=none -build-cache C:\Users\TEST~1.STU\AppData\Local\Temp\arduino_cache_898305 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\examples\01.Basics\Blink\Blink.ino
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
fatal: morestack on g0
... 400+ lines of this.

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\arduino-builder returned 1,073,741,855

Error compiling for board Arduino/Genuino Uno.

facchinm, 2018-10-10:
Hi @Rashantha,
I installed your exact same version in a VM (both 32 and 64 bit) and couldn't reproduce the issue. I think it's related with some other program (antivirus) acting in the background.
Just for reference, a `go` bug exists (https://github.com/golang/go/commit/52e782a2d6e83065e394d127ea5df20e4aaaa8af) but you are the first reporter of this problem, so I doubt it's widespread.
Leaving the issue open of reference; updating to go 1.11.1 **could** fix the issue but we would lose Windows XP compatibility.