arduino/arduino-builder #306: Stops compiling and throws unrelated errors

holgerlembke, 2018-11-14:
We were writting some esp8266 related code, nothing special, just basis stuff. All in all around 600 lines.

Adding and unused and empty function
```
void setupSSDP(void) {
}
```
results in an "error: 'File' was not declared in this scope" at an totally unrelated location.

Changing the name to setuPSSDP makes it compile again, i fact changing the part "setup" to anything else makes it compile again.

You can find the stripped down source code at http://www.lembke.eu/basis2.zip, it can be used with board "Arduino Uno".

facchinm, 2018-11-14:
Hi @holgerlembke ,
this was probably fixed by https://github.com/arduino/arduino-builder/pull/294 , which we didn't integrate yet into any java IDE release.
Would you mind checking if replacing the `arduino-builder` binary with that release solves your issue?

holgerlembke, 2018-11-14:
Checked. Does not fix.

holgerlembke, 2018-11-14:
Rechecked, works fine.

(294 + 304)

closed 2018-11-14 by holgerlembke