arduino/arduino-builder #309: Hardcoded /bin/avr-g++ path

ChrisCummins, 2019-01-13:
On macOS, using the `arduino-1.8.8-macosx.zip` arduino release, compilation of files using arduino-builder fails:

```
$ /Applications/Arduino.app/Contents/Java/arduino-builder \
  -hardware='/Applications/Arduino.app/Contents/Java/hardware' \
  -tools '/Applications/Arduino.app/Contents/Java/tools/' \
  -fqbn 'arduino:avr:uno' \
  /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
fork/exec /bin/avr-g++: no such file or directory
```

System Integrity Protection prevents creating the required symlink:

```
$ sudo ln -s /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ /bin/avr-g++
ln: failed to create symbolic link '/bin/avr-g++': Operation not permitted
```

Version:

```
$ /Applications/Arduino.app/Contents/Java/arduino-builder -version
Arduino Builder 1.4.1
Copyright (C) 2015 Arduino LLC and contributors
See https://www.arduino.cc/ and https://github.com/arduino/arduino-builder/graphs/contributors
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

gvarisco, 2019-01-14:
Thanks for reporting it, Chris. I can indeed reproduce it. 

We shall probably create AND use `/usr/local/bin`, @facchinm as it is already in the default `PATH`.

facchinm, 2019-01-14:
@ChrisCummins wow, we are not even trying to symlink `avr-g++` to `/bin` (this would be madness :slightly_smiling_face:  ).
Let me coordinate with @gvarisco to reproduce it and understand what is going on.
In the meantime, could you test the latest hourly version on the IDE and try to reproduce there? Thanks for reporting!

ChrisCummins, 2019-01-14:
Hi guys, running the same command using the `January 2019 16:29:17 GMT` hourly build from [here](https://www.arduino.cc/en/Main/Software#hourly) produces the same error:

```
$ ~/Downloads/Arduino.app/Contents/Java/arduino-builder -debug-level 10 -hardware ~/Downloads/Arduino.app/Contents/Java/hardware -tools ~/Downloads/Arduino.app/Contents/Java/tools/ -fqbn 'arduino:avr:uno' ~/Downloads/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino

Ts: 1547461917 - Running: GenerateBuildPathIfMissing
Setting build path to /var/folders/dc/_qx355px3dl1gqs1crp5nrjw0000gp/T/arduino-sketch-B685043F11A95C3BD7E78945DC5DF53F
Ts: 1547461917 - Running: EnsureBuildPathExists
Ts: 1547461917 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1547461917 - Running: AddAdditionalEntriesToContext
Ts: 1547461917 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1547461917 - Running: HardwareLoader
Ts: 1547461917 - Running: PlatformKeysRewriteLoader
Ts: 1547461917 - Running: RewriteHardwareKeys
Ts: 1547461917 - Running: ToolsLoader
Ts: 1547461917 - Running: TargetBoardResolver
Ts: 1547461917 - Running: AddBuildBoardPropertyIfMissing
Ts: 1547461917 - Running: LibrariesLoader
Ts: 1547461917 - Running: SketchLoader
Ts: 1547461917 - Running: SetupBuildProperties
Ts: 1547461917 - Running: LoadVIDPIDSpecificProperties
Ts: 1547461917 - Running: SetCustomBuildProperties
Ts: 1547461917 - Running: AddMissingBuildPropertiesFromParentPlatformTxtFiles
Ts: 1547461917 - Running: ContainerBuildOptions
Ts: 1547461917 - Running: CreateBuildOptionsMap
Ts: 1547461917 - Running: LoadPreviousBuildOptionsMap
Ts: 1547461917 - Running: WipeoutBuildPathIfBuildOptionsChanged
Ts: 1547461917 - Running: StoreBuildOptionsMap
Ts: 1547461917 - Running: WarnAboutPlatformRewrites
Ts: 1547461917 - Running: RecipeByPrefixSuffixRunner
Looking for recipes like recipe.hooks.prebuild*.pattern
Ts: 1547461917 - Running: ContainerMergeCopySketchFiles
Ts: 1547461917 - Running: SketchSourceMerger
Ts: 1547461917 - Running: SketchSaver
Ts: 1547461917 - Running: AdditionalSketchFilesCopier
Ts: 1547461917 - Running: loggerAction
Ts: 1547461917 - Running: ContainerFindIncludes
Ts: 1547461917 - Running: PrintUsedAndNotUsedLibraries
Ts: 1547461917 - Running: PrintUsedLibrariesIfVerbose
Ts: 1547461917 - Running: ExportProjectCMake
Ts: 1547461917 - Running: Sizer
fork/exec /bin/avr-g++: no such file or directory
*os.PathError fork/exec /bin/avr-g++: no such file or directory
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/i18n/errors.go:19 (0x1146359)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/utils/utils.go:310 (0x1166e57)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/gcc_preproc_runner.go:63 (0x11842f0)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/container_find_includes.go:333 (0x117ed1b)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/container_find_includes.go:152 (0x117c87d)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:207 (0x117b665)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:125 (0x117abc7)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:223 (0x117ba15)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/arduino-builder/main.go:396 (0x143bc0d)
/home/jenkins/go1.10.2-patched/src/runtime/proc.go:198 (0x102abb2)
/home/jenkins/go1.10.2-patched/src/runtime/asm_amd64.s:2361 (0x1056c11)
```

I'm new to this tool. Perhaps there's some additional PATH arg that I could be setting?

Cheers,
Chris

per1234, 2023-12-27:
The cause of the error was a missing `-tools` flag in the `arduino-builder` command:

```text
-tools '/Applications/Arduino.app/Contents/Java/hardware/tools/avr'
```

The result was the `runtime.tools.avr-gcc.path` property reference here expanding to an empty string:

https://github.com/arduino/ArduinoCore-avr/blob/1.6.23/platform.txt#L21

```text
compiler.path={runtime.tools.avr-gcc.path}/bin/
```

So the conclusion that the error was caused by a hardcoded `/bin/avr-g++` path somewhere was incorrect.

closed 2023-12-27 by per1234