arduino/arduino-builder #311: exec: "/bin/avr-g++": file does not exist (Windows)

bubblegumsoldier, 2019-02-06:
Hey guys,

I was trying to use the arduino-builder to compile my ino file. When I build I receive the error:

`exec: "/bin/avr-g++": file does not exist`

What is strange is that in other threads the error at least contained the full path, in this case the error only contains the /bin/avr-g++. I am using windows. The command I executed looks like this:
` D:\\arduino-nightly\\arduino-builder.exe -debug-level 10 -fqbn 'arduino:avr:nano' -libraries 'D:\\\\arduino-nightly\\libraries' -hardware 'D:\\\\arduino-nightly\\hardware' -tools 'D:\\\\arduino-nightly\\tools' "D:\\\\arduino-nightly\\controller_min.ino"`

The debugging output looks like this:

```
Ts: 1549458475 - Running: GenerateBuildPathIfMissing
Setting build path to C:\Users\HENRYM~1\AppData\Local\Temp\arduino-sketch-BEC378AA16B241A14DEA1E1E0D814081
Ts: 1549458475 - Running: EnsureBuildPathExists
Ts: 1549458475 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1549458475 - Running: AddAdditionalEntriesToContext
Ts: 1549458475 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1549458475 - Running: HardwareLoader
Ts: 1549458475 - Running: PlatformKeysRewriteLoader
Ts: 1549458475 - Running: RewriteHardwareKeys
Ts: 1549458475 - Running: ToolsLoader
Ts: 1549458475 - Running: TargetBoardResolver
Ts: 1549458475 - Running: AddBuildBoardPropertyIfMissing
Ts: 1549458475 - Running: LibrariesLoader
Ts: 1549458475 - Running: SketchLoader
Ts: 1549458476 - Running: SetupBuildProperties
Ts: 1549458477 - Running: LoadVIDPIDSpecificProperties
Ts: 1549458477 - Running: SetCustomBuildProperties
Ts: 1549458477 - Running: AddMissingBuildPropertiesFromParentPlatformTxtFiles
Ts: 1549458477 - Running: ContainerBuildOptions
Ts: 1549458477 - Running: CreateBuildOptionsMap
Ts: 1549458477 - Running: LoadPreviousBuildOptionsMap
Ts: 1549458477 - Running: WipeoutBuildPathIfBuildOptionsChanged
Ts: 1549458477 - Running: StoreBuildOptionsMap
Ts: 1549458477 - Running: WarnAboutPlatformRewrites
Ts: 1549458477 - Running: RecipeByPrefixSuffixRunner
Looking for recipes like recipe.hooks.prebuild*.pattern
Ts: 1549458477 - Running: ContainerMergeCopySketchFiles
Ts: 1549458477 - Running: SketchSourceMerger
Ts: 1549458477 - Running: SketchSaver
Ts: 1549458477 - Running: AdditionalSketchFilesCopier
Ts: 1549458479 - Running: loggerAction
Ts: 1549458479 - Running: ContainerFindIncludes
Ts: 1549458479 - Running: PrintUsedAndNotUsedLibraries
Ts: 1549458479 - Running: PrintUsedLibrariesIfVerbose
Ts: 1549458479 - Running: ExportProjectCMake
Ts: 1549458479 - Running: Sizer
exec: "/bin/avr-g++": file does not exist
*exec.Error exec: "/bin/avr-g++": file does not exist
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/i18n/errors.go:19 (0x530b4a)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/utils/utils.go:310 (0x54bb3a)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/gcc_preproc_runner.go:63 (0x563ac1)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/container_find_includes.go:333 (0x55f4be)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/container_find_includes.go:152 (0x55d58b)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:207 (0x55c711)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:125 (0x55bd1a)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:223 (0x55ca09)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/arduino-builder/main.go:396 (0x798e5f)
/home/jenkins/go1.10.2-patched/src/runtime/proc.go:198 (0x42a0ba)
/home/jenkins/go1.10.2-patched/src/runtime/asm_386.s:1665 (0x451221)
```

Thanks in advance

facchinm, 2019-02-07:
Hi @bubblegumsoldier ,
the builder is not finding the correct tool since it's embedded in an IDE installation (aka bundled).
Tools for avr are located in `$IDE/hardware/tools/avr`, so your commandline should become

```
`D:\\arduino-nightly\\arduino-builder.exe -debug-level 10 -fqbn 'arduino:avr:nano' -libraries 'D:\\\\arduino-nightly\\libraries' -hardware 'D:\\\\arduino-nightly\\hardware' -tools 'D:\\\\arduino-nightly\\tools-builder' -tools 'D:\\\\arduino-nightly\\hardware\\tools\\avr' "D:\\\\arduino-nightly\\controller_min.ino"
```

Let me know if it works!

bubblegumsoldier, 2019-02-07:
@facchinm That actually kind of worked, it shows a different error now. It sais

```
avr-g++: error: missing device or architecture after '-mmcu='
```

in full:

```
Ts: 1549533836 - Running: GenerateBuildPathIfMissing
Setting build path to C:\Users\HENRYM~1\AppData\Local\Temp\arduino-sketch-E9C550D5338228309B49D90AD705EBD8
Ts: 1549533836 - Running: EnsureBuildPathExists
Ts: 1549533836 - Running: ContainerSetupHardwareToolsLibsSketchAndProps
Ts: 1549533836 - Running: AddAdditionalEntriesToContext
Ts: 1549533836 - Running: FailIfBuildPathEqualsSketchPath
Ts: 1549533836 - Running: HardwareLoader
Ts: 1549533836 - Running: PlatformKeysRewriteLoader
Ts: 1549533836 - Running: RewriteHardwareKeys
Ts: 1549533836 - Running: ToolsLoader
Ts: 1549533836 - Running: TargetBoardResolver
Ts: 1549533836 - Running: AddBuildBoardPropertyIfMissing
Ts: 1549533836 - Running: LibrariesLoader
Ts: 1549533836 - Running: SketchLoader
Ts: 1549533837 - Running: SetupBuildProperties
Ts: 1549533837 - Running: LoadVIDPIDSpecificProperties
Ts: 1549533837 - Running: SetCustomBuildProperties
Ts: 1549533837 - Running: AddMissingBuildPropertiesFromParentPlatformTxtFiles
Ts: 1549533837 - Running: ContainerBuildOptions
Ts: 1549533837 - Running: CreateBuildOptionsMap
Ts: 1549533837 - Running: LoadPreviousBuildOptionsMap
Ts: 1549533837 - Running: WipeoutBuildPathIfBuildOptionsChanged
Build options changed, rebuilding all
Ts: 1549533838 - Running: StoreBuildOptionsMap
Ts: 1549533838 - Running: WarnAboutPlatformRewrites
Ts: 1549533838 - Running: RecipeByPrefixSuffixRunner
Looking for recipes like recipe.hooks.prebuild*.pattern
Ts: 1549533838 - Running: ContainerMergeCopySketchFiles
Ts: 1549533838 - Running: SketchSourceMerger
Ts: 1549533838 - Running: SketchSaver
Ts: 1549533838 - Running: AdditionalSketchFilesCopier
Ts: 1549533840 - Running: loggerAction
Ts: 1549533840 - Running: ContainerFindIncludes
Ts: 1549533840 - Running: FailIfImportedLibraryIsWrong
Ts: 1549533840 - Running: WarnAboutArchIncompatibleLibraries
Ts: 1549533840 - Running: loggerAction
Ts: 1549533840 - Running: PreprocessSketch
Ts: 1549533840 - Running: ContainerAddPrototypes
avr-g++: error: missing device or architecture after '-mmcu='
Ts: 1549533840 - Running: PrintUsedAndNotUsedLibraries
Ts: 1549533840 - Running: PrintUsedLibrariesIfVerbose
Ts: 1549533840 - Running: ExportProjectCMake
Ts: 1549533840 - Running: Sizer
exit status 1
*exec.ExitError exit status 1
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/i18n/errors.go:19 (0x530b4a)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/utils/utils.go:323 (0x54bac3)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/gcc_preproc_runner.go:49 (0x5639bd)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/container_add_prototypes.go:54 (0x55cc7f)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:207 (0x55c711)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:154 (0x55c1e6)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:207 (0x55c711)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:125 (0x55bd1a)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/builder.go:223 (0x55ca09)
/home/jenkins/workspace/arduino-builder-all-cross/src/github.com/arduino/arduino-builder/arduino-builder/main.go:396 (0x798e5f)
/home/jenkins/go1.10.2-patched/src/runtime/proc.go:198 (0x42a0ba)
/home/jenkins/go1.10.2-patched/src/runtime/asm_386.s:1665 (0x451221)
```

facchinm, 2019-02-07:
It happens because you are targeting a nano, so you must specify the cpu model (see 
https://github.com/arduino/ArduinoCore-avr/blob/5755ddea49fa69d6c505c772ebee5af5078e2ebf/boards.txt#L143
Either `arduino:avr:nano:cpu=atmega328`,  `arduino:avr:nano:cpu=atmega328old` and  `arduino:avr:nano:cpu=atmega168` are available, so you can choose the one that fits your board.

closed 2019-02-07 by facchinm

Givy, 2022-01-23:
Hi,
Is this the same issue for me?  I followed an advice on the forum and everything fail apart after I removed "arduino15".  Original issue was "error on avrdude...".  Please advise.  Thanks.

Detecting libraries used...
"C:\\Users\\Engineer\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10820 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Engineer\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\cores\\arduino" "-IC:\\Users\\Engineer\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\variants\\mega" "C:\\Users\\Engineer\\AppData\\Local\\Temp\\arduino_build_583993\\sketch\\ds3231.ino.cpp" -o nul

exec: "C:\\Users\\Engineer\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++": file does not exist
Error compiling for board Arduino Mega or Mega 2560.