arduino/arduino-builder #312: Error Message with Multiple Blinks

YesmindaHeganli, 2019-02-20:
I keep on getting this message;
WARNING: library Scheduler claims to run on [sam architecture(s) and may be incompatible with your current board which runs on samd] architecture(s).
C:\Users\cf3504\AppData\Local\Temp\cc4eUy6s.s: Assembler messages:

C:\Users\cf3504\AppData\Local\Temp\cc4eUy6s.s:46: Error: constant value required

exit status 1
Error compiling.
; whenever I try to program my arduino board with this program;
https://www.arduino.cc/en/Tutorial/MultipleBlinks
; Can someone please help?

YesmindaHeganli, 2019-02-20:
[Code Error Example.zip](https://github.com/arduino/arduino-builder/files/2885911/Code.Error.Example.zip)

This is what it currently looks like, for those who would need it.

facchinm, 2019-02-21:
Hi @YesmindaHeganli ,
it looks like you are using an outdated [Scheduler](https://github.com/arduino-libraries/Scheduler) library.
Download the new one from the library manager and the problem should be gone.

closed 2019-02-21 by facchinm

YesmindaHeganli, 2019-02-21:
@facchinm There isn't another version of scheduler in manage libraries. Should I download it from the internet?

facchinm, 2019-02-22:
Latest version available via library manager should be 0.4.4 , and it's the same that's available on github. Would you mind checking your version?

YesmindaHeganli, 2019-02-22:
I do have 0.4.4 when I check it.

facchinm, 2019-03-04:
Can you please attach the full verbose compilation after turning it on from IDE File/Preferences?